ArduPilot/ardupilot #20678: sim_vehicle.py   - fail to forward "--cmd=" arguments to MAVproxy

AndKe, 2022-05-02:
## Bug report

**Issue details**
The sim_vehicle.py mishandles the --cmd parameter, expected to be forwarded to mavproxy

this command line ```  sim_vehicle.py -v ArduCopter -f hexa -C --cmd="module load checklist;"```
is expected to load the checklist module.
instead, it ends up as multiple arguments as the quotation marks get removed

**Version**
current master

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n/a

**Hardware type**
n/a

**Logs**
```
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--cmd" "module" "load" "checklist;"
RiTW: Starting ArduCopter : /home/andre/prog/ardupilot/build/sitl/bin/arducopter -S --model hexa --speedup 1 --slave 0 --defaults prog/ardupilot/Tools/autotest/default_params/copter.parm,prog/ardupilot/Tools/autotest/default_params/copter-hexa.parm -I0 --home 69.775839,19.548928999999987,87.0,170.0
ERROR: mavproxy takes no position arguments; got (['load', 'checklist;'])
SIM_VEHICLE: MAVProxy exited
SIM_VEHICLE: Killing tasks

```

khanasif786, 2022-05-11:
Instead You can use ```sim_vehicle.py -v ArduCopter -f hexa -C --load-module=checklist```

AndKe, 2022-05-11:
@khanasif786 Thank you! - that works fine.  
That's a nice workaround, I guess the behavior I caused is still a bug?

khanasif786, 2022-05-11:
Actually, the space separated quoted commands is not handled (implemented) anywhere in the sim_vehicle.py. Maybe because there are another workarounds like that I mentioned (for loading modules). But I also think it should directly pass those commands that are present in a quoted text without separating that into different list, **so the bug exists**. Maybe it should be seen more like a feature to support those commands idk ¯\_(ツ)_/¯.