ArduPilot/ardupilot #20711: Some parameter range is not correctly checked in source code

Czw007, 2022-05-07:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**
Some parameter range is not checked in source code. For example , the parameter "ATC_ANG_RLL_P"，For example, the parameter "ATC_ANG_RLL_P", the range specified in the user manual is 3.000 - 12.000, but setting it to a value outside the range can also be accepted by the program. Maybe this is a program bug that does not properly check the range of configuration parameters.

**Version**
Copter beta V4.2.0 rc1

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

hendjoshsr71, 2022-05-07:
Unfortunately, except for very specific parameters (major safety reasons) we cannot currently perform checks on every parameter.
The reason is because the hardware doesn't have enough flash space available for these types of checks. 

However, I have been considering creating a LUA script to encompass these sorts of checks during tuning or flight testing.

rmackay9, 2022-05-08:
This is also partially on purpose.  The parameter ranges are for the ground station to enforce but the flight code will accept values outside this which is sometimes necessary for vehicles that are much larger or smaller than the norm.

I think I might close this because it's not really a bug and I'm not sure we would even want to check all parameters.  There are probably more checks that we could add for very obvious problems (like negative values).

closed 2022-05-08 by rmackay9