ArduPilot/ardupilot #20713: kakutef7mini sbus rc input do not work

NSG-666, 2022-05-08:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

1.I use a AC900 receiver for my kakutef7mini, and set the RC Input follow below step(select the sbus way), it do not work:
[image]

2.So i try to go back to the old RC input way, i change the hwdef.h to use EICU to deal the RC input, it work good:
[image]
[image]

**Version**
ardupilot rover 4.1.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[*] Rover
[  ] Submarine

**Airframe type**
rover undefined

**Hardware type**
kakutef7mini

**Logs**
debug screen shot.

Vkd4U2NtVkZOVg, 2022-05-08:
Hi @NSG-666, May I ask you the configuration about the online debug with the cubeide? Is there a way to contact you? maybe a wechat?

NSG-666, 2022-05-08:
> Hi @NSG-666, May I ask you the configuration about the online debug with the cubeide? Is there a way to contact you? maybe a wechat?

I just use it to debug the elf file.

rmackay9, 2024-04-20:
I know this is a very old report but we have made some changes to the SBUS support in 4.5.x.  I wonder if you'd be willing to give it a try?

peterbarker, 2024-07-31:
Ping @NSG-666 - can you test 4.5.x please?

peterbarker, 2024-11-24:
Closing this one - no response from author.

Would be nice to get feedback.  Please re-open if problems are still present on 4.5.x (or the 4.6 beta)

closed 2024-11-24 by peterbarker