ArduPilot/ardupilot #20716: AP_SmartRTL algorithm breaks on Circle mode

peterbarker, 2022-05-09:
## Bug report

**Issue details**

A flight on the weekend showed a problem where the SmartRTL buffer filled up while the vehicle was in circle mode, and SmartRTL disengaged.

This is trivially reproducible in SITL:
```
mode loiter
arm throttle
rc 3 2000
mode circle
rc 3 1500
set sim_speedup 100
```

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Chimera

**Hardware type**
Lumeneer H743

**Logs**
n/a - reproducible in SITL

rmackay9, 2022-05-09:
@peterbarker,

It's not really a bug though is it?  It's just running out of space in the buffer right?

muramura, 2022-05-17:
This phenomenon has existed since SMARTRTL was released.
I saw the same message that I could not SMARTRTL in the same way, in circle mode, when I let it rise.

peterbarker, 2024-07-31:
> @peterbarker,
> 
> It's not really a bug though is it? It's just running out of space in the buffer right?

I believe it is - one of the phases should be cutting the loops out so you'll never grow past a certain point.

This is still reproducible on master.

rmackay9, 2024-07-31:
I'm pretty certain that it is cutting loops so I suspect it is that the loops are too small or the vehicle's altitude is drifting