ArduPilot/ardupilot #20737: Possibilty to use OptFlow pointing up

simon88, 2022-05-11:
## Feature request

be able to use an optical sensor (HereFlow example) pointing up. I think it would be nice to be able to use an upward facing optflow sensor, because in the case of full indoor flight there is a lot more grip on the ceiling than on the floor.

**Describe the solution you'd like**
I think that an additional parameter to indicate if the sensor is at the bottom or at the top would be suffisant.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2022-05-11:
Hi @simon88,

I've actually created a PR that adds this functionality to the EKF3 but it needs testing to confirm I haven't made any mistakes.
https://github.com/ArduPilot/ardupilot/pull/20038

I'm pretty sure there is at least one mistake in the PR which is that the FLOW_ORIENT parameter needs to be set to 8 or 12 (sorry I forget which) for it to work correctly.

simon88, 2022-05-11:
Hi @rmackay9 

Ah nice this is a great news, thanks, so I closed this request :)

closed 2022-05-11 by simon88