ArduPilot/ardupilot #20754: Plane Auto TakeOff : Idle Throttle for Hand/Bungee/Catapult launch

romarysonrier, 2022-05-13:
## Feature request

**ESC may delay startup when using folding propeler or heavy propeller**
If the AUW is near the max supported by a airframe, delayed spinning will cause crash

**My proposal is to implement a param TKOFF_IDLE_MODE with 3 values possibles:**
MODE=0 throttle when idle is equal to 0%
MODE=1 throttle when idle is equals to param TKOFF_IDLE_THR (static value, in % of motor PWM) . The motor should start to spin when RC MOTOR INPUT >1500 for safety reason.
MODE=2 throttle equals RC MOTOR INPUT (eg channel 3), limited by the upper value define by TKOFF_IDLE_THR (in % of motor PWM). My prefered mode.

So basicaly, MODE=0 is the current behaviour, MODE=1 is the INAV feature, MODE=2 allow the pilot to take manual control with its RC on the motor startup, with a predefined “security” limit .

To sumup my main feature request is to have a “balanced folding propeller” BEFORE launch time… and the only known way is to have a 10-30% throttle when the plane is IDLE. INAV have this feature for this specific prupose..

**Known Alternative : FBWA**
_But FBWA launch requires at least 2 people to be securly executed., And it is not as convenient as AUTO TakeOFF._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Take-off is kind of **very risky** with that kind of frame using arduplane Auto-Takeoff because if motor is not spinning before launching the plane (Idle Throttle is equal to 0% by default), there is a high risk that the ESC has to retry its “startup procedure” due to the torque required at startup-time ( if propeler is folded or unbalanced) . causing the propeler to have a 1-2 sec delay before really spinning… leading to some crashes… especially with wings that have fewer attitude authority than t-tail plane at low speed.

What’s-more if the propeller itself is heavy or just big, the torque required to speed-up the motor will generate some roll at launch-time, what is very unwanted at takeoff, causing loss of lift… leading to some crashes…

[INAV has a parameter for that issue](https://inavfixedwinggroup.com/guides/tuning/how-to-setup-nav-launch-correctly/#6-nav_fw_launch_idle_thr) : nav_fw_launch_idle_thr
Here is the associated documentation for folding propeler :

It can be very stressful for your motor, props and even fuselage if a folding propeller is spinning instantly to 100% power if it is folded. So, you can use a low value like 1100 to make sure the prop is open and in balance before launch.

Most of my wing build (Z89, SonicModel HD Wing, AR Wing Pro) are using big folding propeller, that fails to start is most of case if not spinning.

peterbarker, 2022-05-14:
FWIW, my immediate thoughts on this were a "shake to wake" thing where it would go to some minimal throttle level, then waiting for a groundspeed to go to takeoff throttle.

No idea what happens in the case that the thing screws up going to that minimal throttle level...

romarysonrier, 2022-05-14:
Yes , the basic idea would be to to have a minimal throttle level before launching the plane in auto takeoff mode. 

As the changes should only impact autotakeoff, which is already an computer-assisted takeoff, so i guess pilot have to assume there is some risks... 

By the way, i have faced the folding propeller issue ( causing esc  to stop and start its startup procedure) may times. And as far I remember, the crash detection failed to stop the motor. I guess the reason is this feature is limited to the AUTO mode...  may be we should consider to embedded this feature to the AUTOTAKEOFF mode,  since with this TKOFF_IDLE_MODE>0 , the propeller will be spinning for sure...

⁰

suraqis, 2022-05-15:
THR_SUPP_MAN = 1 enables manual throttle input in Auto Takeoff before launch acc and speed thresholds hit.

romarysonrier, 2022-05-15:
Does this issue is still applicable for arduplane 4.2 ?https://github.com/ArduPilot/ardupilot/issues/12453

My esc have breaking enable, so propeler might get unfold & unbalanced....

robustini, 2023-02-17:
This was very interesting, personally for hand launching I prefer the iNav procedure, much saferin the absence of airdpeed and for vertical launches.
It would be nice if in Ardupilot they would evolve the hand launch procedure.

Georacer, 2024-11-13:
@robustini  @romarysonrier perhaps you'd want to take a look at #28351.

robustini, 2024-11-14:
> @robustini @romarysonrier perhaps you'd want to take a look at #28351.

Thank you very much, although it doesn't cover conceptually everything that has been asked, but it is something.