ArduPilot/ardupilot #20767: Custom Build Server Enhancements

peterbarker, 2022-05-15:
## Feature request

Our Custom Build Server (https://custom.ardupilot.org/) is great, but is lacking a lot of features which would make it very useful.  Given that when adding one feature it is often convenient to add several, this is a compendium issue of features which people have mentioned as being potentially useful.

 - ability to select a branch to build (instead of "only master")
    - this will require inspecting the selected branch to work out which option are available to present to the user
    - selecting branches from a white-listed github account name list would be a useful follow-up to this
 - ability to modify interface based on which vehicle is to be built
 - ability to build AP_Periph (linked to above point)
 - ability to upload a hwdef.dat (and perhaps other files) to populate the interface.  We already allow downloading of the hwdef.dat
 - ability to specify default parameter values to be included in the build
 - ability to specify some parameters as read-only
 - ability to enable assertions

tridge, 2022-05-23:
for periph it would be nice to have string input for node name, and integers for default ports and backends

hendjoshsr71, 2022-09-27:
- Add a select all button for everything in a group

- Add a GUI frontend to allow for common "packages" of features by picture for example.

Hwurzburg, 2023-02-09:
aside from the defaults being populated when a board/vehicle/branch is chosen, the ability to collapse/expand all boxes is really needed in order to facilitate changes...the two biggest complaints I get from users...

adityaomar3, 2023-09-10:
Can somebody help me with what parameters are to be added and its significance ? I am interested in the issue.

Theonewhomadethings, 2024-03-26:
Is there still room to contribute on this issue?, looking for a first issue to be assigned.  @peterbarker

shiv-tyagi, 2024-03-26:
Hey @Theonewhomadethings! Thanks for reaching out. Feel free to take up anything from the list posted above. Let me know if you need help with anything.