ArduPilot/ardupilot #20768: Neopixel not working with plane 4.2

Vkd4U2NtVkZOVg, 2022-05-15:
## Bug report
worked with 4.1.7, worked with master but not 4.2.0
**Issue details**
It's not flashing the correct color
Please describe the problem_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ x ] Rover
[  ] Submarine

MallikarjunSE, 2022-05-15:
https://github.com/ArduPilot/ardupilot/issues/20583
Similar?

andyp1per, 2022-05-15:
Does it work on 4.3-dev?

andyp1per, 2022-05-15:
Incidentally I tested this on an omnibus F4 pro (F405) and it worked fine for me

Vkd4U2NtVkZOVg, 2022-05-15:
> Does it work on 4.3-dev
Yes, It is worked with 4.3-dev, could you please tell me which pr fixed it? maybe I can fixed it for myself?

Vkd4U2NtVkZOVg, 2022-05-15:
> #20583 Similar?
Yes, The led is flashing, but the color is wrong, and It seems only on the STM32F4. I test it with F7 was fine.

andyp1per, 2022-05-15:
You could try this: https://github.com/ArduPilot/ardupilot/pull/20432 it's not the PR I was thinking of, but might be worth trying

Vkd4U2NtVkZOVg, 2022-05-15:
@andyp1per  thanks a lot! after apply the change from the pr you mentioned, the led works right.

andyp1per, 2022-05-15:
@rmackay9 FYI

closed 2022-05-15 by Vkd4U2NtVkZOVg

rmackay9, 2022-05-15:
ok noted.. we can try and get this into Copter/Rover-4.2.1.  I think the boat has sailed on 4.2.0 unless something really serious turns up.