ArduPilot/ardupilot #20784: AP_Param: Setting the maximum value of AP_INT32 causes SIGFPE exception

muramura, 2022-05-17:
## Bug report

**Issue details**
_Please describe the problem_

In SITL, I set BRD_SERIAL_NUM to the maximum AP_INT32 value of 2147483647, which caused a SIGFPE exception.

**Version**
_What version was the issue encountered with_

 ArduCopter V4.3.0-dev (1090a881) 

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

Copter

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

GDB
[image]

GCS
[image]

hendjoshsr71, 2022-05-18:
Anything above 24-bits is not guaranteed to work.  

See the comment here from the PR of yours. https://github.com/ArduPilot/ardupilot/pull/19854#issuecomment-1020276604

I did check on CubeOrange with INT32_MAX and of course the 2^24 limits set in the new parameter description.  Neither crashed.   

For the SITL side I only had issues very close to INT32_MAX.

Not sure what we can do about it?

nttedt-murata, 2022-05-18:
ArduPilot suppresses the STM32 exception.

muramura, 2022-05-18:
ardupilot is suppressing stm32 exceptions and will continue processing.  The value of the explanation was ok for me too.