ArduPilot/ardupilot #20816: Copter-4.2.0 - Possible problem found that doesn't make the leds work normally (DShot) (luminousbees)

DavideLentini, 2022-05-22:
**Issue details**

Recently my team and I have been testing various "beta" versions of ArduCopter 4.2.0 , which include (rc1-rc2-rc3-rc4) . We tested these versions on a LuminousBee drone which has 100 leds . Normally we test these leds through a custom firmware that is still based on the Copter-4.2 branch . By testing them with 4.2.0-rc1 , **we immediately noticed a problem in turning on the leds or otherwise abnormal behavior . In fact, of the 100 leds that were normally supposed to turn on , only 25 turned on** . To be sure that it was not a problem with the drone , we compared the behavior of the leds between ArduCopter 4.1.5 and ArduCopter 4.2.0-rc1 on the same drone  , noting that in 4.1.5 all the leds light up correctly . 

Intuiting then that the problem could be found in 4.2.0-rc1 , we also tested later versions up to rc4 , noting that the problem occurred in those as well .

After a lengthy analysis of the commits , using the `git bisect tool` , we found what might be causing the problem in turning on the leds .

The problem ( in our case ) is found in the file :

`/libraries/AP_HAL_ChibiOS/RCOutput.cpp`

And more precisely in 3 lines of code :

https://github.com/ArduPilot/ardupilot/blob/cf4768454f55f4d9753945ddc3507b03db57ccb6/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L255

https://github.com/ArduPilot/ardupilot/blob/cf4768454f55f4d9753945ddc3507b03db57ccb6/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L256

https://github.com/ArduPilot/ardupilot/blob/cf4768454f55f4d9753945ddc3507b03db57ccb6/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L257

In fact, we noticed that by marking a specific commit as "bad" , the LEDs would go back to working all 100 . 

We then decided ( to temporarily curb the problem ) to comment out the 3 lines of code reported above in the rc4 version and surprisingly we were able to get all 100 leds working . 

So I think our problem is purely related to these 3 lines added ( or modified ) in this commit : 

https://github.com/ArduPilot/ardupilot/commit/4c977eb7a0888602ced512bf81a2ae11645058ed

Which is exactly what we marked as bad , to overcome the problem to the leds .

So this is what I would like to report , in the hope that this problem can be fixed in the official Copter 4.2.0 release .

I would like to report this issue directly to @tridge (and @ntamas who helped me ) , since he was the one who made the commit concerned

Thanks to  all

**Version**
4.2.0(rc1-rc2-rc3-rc4)

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

ntamas, 2022-05-22:
Just to add my two cents: the issue was observed with the stock ArduCopter firmware (4.2.0-rc1 onwards, latest beta is not affected) so it has nothing to do with our patching.

The LEDs are NeoPixel LEDs, driven from a PWM pin.

andyp1per, 2022-05-22:
Thanks for the report. If you change https://github.com/ArduPilot/ardupilot/blob/cf4768454f55f4d9753945ddc3507b03db57ccb6/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L255 is there a (bigger) value for which things start working?

andyp1per, 2022-05-22:
Also please post the parameters you are using

DavideLentini, 2022-05-22:
> Thanks for the report. If you change
> 
> https://github.com/ArduPilot/ardupilot/blob/cf4768454f55f4d9753945ddc3507b03db57ccb6/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L255
> 
> is there a (bigger) value for which things start working?
> For instance try multiples of `_dshot_period_us`

Thank you , for the answer and the advice . I will have a chance to share with you the parameters used by us and test your advice in the day tomorrow , so stay tuned !

DavideLentini, 2022-05-23:
> Thanks for the report. If you change
> 
> https://github.com/ArduPilot/ardupilot/blob/cf4768454f55f4d9753945ddc3507b03db57ccb6/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L255
> 
> is there a (bigger) value for which things start working?
> For instance try multiples of `_dshot_period_us`

Hi , today i am starting by testing your advice , i have one question : When you suggesting to try a bigger value of actually _dshot_period_us ( three times ) , can i actually change the values in thise line : 

 const uint32_t max_delay_us = _dshot_period_us;  
to
 const uint32_t max_delay_us = _dshot_period_us * 3 ; 

This change shoud be work , right ?

andyp1per, 2022-05-23:
Yes, exactly that

DavideLentini, 2022-05-23:
> Yes, exactly that
I did a test by setting the multiplier to * 3 .
With this multiplier about 70 leds out of 100 turn on 
. I then tried increasing it to 5 and the leds all turn on . So I think the right multiplier to turn on all the leds is 5 .
` const uint32_t max_delay_us = _dshot_period_us * 5;  `
Out of curiosity I also did several tests with lower multipliers and I really saw how lowering the multiplier decreased the number of leds turned on .

andyp1per, 2022-05-23:
Hi @DavideLentini - so thanks for confirming my theory - basically the NeoPixel cycle time goes up with the number of LEDs and in your case the timeout is not long enough. However, the default timeout is the maximum cycle available - so for regular setup 1ms (1Khz) - or for high loop rates with SERVO_DSHOT_RATE somewhat lower. So your configuration is currently starving the ESCs of signal when the NeoPixel elements are written - if 5* fixes it then you are losing 4 dshot pulses every time the LEDs get written to. If the pause is longer than 30ms then the motors will disarm, so although you are reasonably far away from that its still not great. I am happy to increase the timeout if @tridge agrees, although supporting 100 LEDs seems pretty arbitrary.

DavideLentini, 2022-05-24:
> Hi @DavideLentini - so thanks for confirming my theory - basically the NeoPixel cycle time goes up with the number of LEDs and in your case the timeout is not long enough. However, the default [...]

Hi @andyp1per I am a bit of a beginner in the field of Dshot , but from what I understand the method we used to turn on the LEDs temporarily is not good because it conflicts with the communication with the ESCs.

 Right now we need ASAP , a 4.2.0 Copter firmware to test , which has none of these problems . So if this problem can be solved by going to change the default timeout by asking @tridge I would say speed up this process. :)

DavideLentini, 2022-05-26:
So after a few days of research, we have discovered that on our drones the DMA channel is shared by both the esc and leds controls. 
We are still investigating how this could be solved, although for now the least intrusive way seems to be to use a multiplier to increase the max_delay_us . 
Instead @andyp1per  have you had a chance to talk to @tridge  to discuss whether it is possible to increase the timeout ? Anyway, could you tell me what particular procedure I need to do to change the timeout so that I can test it without having to manually change the max_delay_us and causing the worsing of the signals to escs . 
Thanks :)

andyp1per, 2022-05-26:
With the current code there is no way of increasing the timeout with affecting the ESCs - for this reason it seems unlikely that we would increase the timeout by default as it could be considered a flight risk. Longer term we need to restructure the code to use a thread-per-timer so that the LED thread can run independently of the ESCs. This wouldn't actually help you though if the DMA channel is shared. So for now I think you will need to increase the timeout manually yourself.

andyp1per, 2022-05-26:
The line we had already discussed

DavideLentini, 2022-05-26:
I understand, thank you. I thought I understood  that manually raising the timeout could cause the motors to disarm, or did I misunderstand? 
_. So your configuration is currently starving the ESCs of signal when the NeoPixel elements are written - if 5* fixes it then you are losing 4 dshot pulses every time the LEDs get written to. If the pause is longer than 30ms then the motors will disarm, so although you are reasonably far away from that its still not great_

Thanks

rmackay9, 2022-05-30:
I'm not 100% following the progress on this issue but I think it may be resolved in [4.2.1-rc1](https://discuss.ardupilot.org/t/copter-4-2-1-rc1-available-for-beta-testing/86125).

@DavideLentini, could you re-test with [4.2.1-rc1](https://discuss.ardupilot.org/t/copter-4-2-1-rc1-available-for-beta-testing/86125) downloaded directly from MP or QGC or confirm?

andyp1per, 2022-05-30:
Hi @rmackay9 - no it won't be resolved, this is a more fundamental problem

DavideLentini, 2022-05-30:
Hello everyone , thanks for the help . I tried a little while ago to test the ArduCopter 4.2.1 - rc1 with several drones and the result is the initial one , only 20 leds always light up .

andyp1per, 2023-11-24:
Fixed by https://github.com/ArduPilot/ardupilot/pull/23019

closed 2023-11-24 by andyp1per