ArduPilot/ardupilot #20830: AP + ROS interface

rmackay9, 2022-05-24:
AP generally uses mavros to interface with the original ROS but we don't have an official interface for ROS2 which is fairly important.

We have two possible solutions:

- This is a mostly working but unmerged port of mavros to ROS2 (https://github.com/uas-at-ucla-dependencies/mavros)
- [PR 17779](https://github.com/ArduPilot/ardupilot/pull/17779) adds a direct AP<->ROS2 interface but is incomplete

amilcarlucas, 2022-05-24:
there is also the ros2 branch at the mavlink/mavros repository.

eMrazSVK, 2022-09-16:
Hello, is there any update on this topic? I am considering contributing to ArduPilot ROS2 support and I am trying to figure out, what would be the best thing to do. I cannot estimate what state is the https://github.com/ArduPilot/ardupilot/pull/17779 in or if its even wanted to port MAVROS to ROS2 without direct DDS support (finisihing this fork for example https://github.com/uas-at-ucla-dependencies/mavros).

Thank you for any information.

Ryanf55, 2023-03-14:
An update. I'm going for fully fledged ROS2 support with PR #17779 which is likely merging soon. 
This will provide support for native ROS2 message types so it works out of the box with gazebo, foxglove, rqt_viz, and other standard ROS2 tooling. There is no dependency on mavlink or mavros or any bridges like that. If it's interesting, feel free to follow up on the ROS discourse thread: 
https://discourse.ros.org/t/microxrce-dds-on-ardupilot/29972
^ TLDR: I'm going to try to get more alignment on ROS REP-147 and accept that as the standard AP+ROS2 interface. 
ROS1 is out of scope of this work.

Ryanf55, 2023-08-08:
I'm closing this, the ROS 2 interface is being worked on, and support for many topics is merged in to `master`, and will be the official interface. It will be part of the 4.5 release, including examples for using Google Cartographer and NAV2.

closed 2023-08-08 by Ryanf55