ArduPilot/ardupilot #20849: Pl update Mavlink submodule, and backport

olliw42, 2022-05-26:
Currently, master, Copter-4.2, and Plane-4.2 link to a somewhat older version of the ArduPilot/mavlink repo. Unfortunately this holds an older version of the storm32xml custom dialect, dated 4. Feb. 2021.

The ArduPilot/mavlink repo is already in much better sync with mavlink upstream (especially https://github.com/ArduPilot/mavlink/commit/94c40f955c71d32b2a9c9236ea10c7d6c292d779), and holds the latest version of storm32.xml, which is dated 6. Okt. 2021.

so, I kindly ask you to update the mavlink submodule, especially for Copter-4.2 and Plane-4.2. It would be great if it could become available with the next releases of Copter & Plane.

Thanks, and cheers, Olli

khancyr, 2022-05-26:
Seems fair. That shouldn't be an hard work to backport on 4.2 serie

rmackay9, 2022-06-07:
I've added this request to the Copter-4.2 issues list and added the backport label to PeterB's PR so it will at least be discussed as to whether we can safely back port it or not (I think we should be able to but we need to discuss.. )

olliw42, 2023-02-03:
closing it since outdated :)

closed 2023-02-03 by olliw42