ArduPilot/ardupilot #20850: Docker permissions issues

zacroker, 2022-05-26:
## Bug report

**Issue details**
Docker has permissions issues. In the wiki the relevant instructions can be found [at this link](https://ardupilot.org/dev/docs/building-setup-linux.html#how-to-build-the-docker-image). 

1. `docker build . -t ardupilot`
2. `docker run --rm -it -v `pwd`:/ardupilot ardupilot:latest bash`

When you are in the container and trying to build etc, you get permissions issues as the user in the docker container is 'ardupilot' with uid/gid 1000. Hence when you volume mount the user/group are the local user/group and unless they are uid/gid 1000 you cannot create the file 'tasklist.json'. See the below log output for an example command in the container.

```
ardupilot@1ca879292c95:/ardupilot/ArduCopter$ sim_vehicle.py -w
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 158, in waf_entry_point
    run_commands()
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 248, in run_commands
    run_command('init')
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "/ardupilot/modules/waf/waflib/Context.py", line 204, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/ardupilot/wscript", line 91, in init
    generate_tasklist(ctx, False)
  File "/ardupilot/wscript", line 565, in generate_tasklist
    with open(os.path.join(Context.top_dir, "tasklist.json"), "w") as tlist:
PermissionError: [Errno 13] Permission denied: '/ardupilot/tasklist.json'
```

**Version**
Latest master

**Proposed Solution**
My proposal for a solution is as follows. The Dockerfile has uid/gid 1000 as default, but they can be overridden by a build argument  e.g.

```
FROM ubuntu:20.04
WORKDIR /ardupilot

ARG UID=1000
ARG GID=1000
...
```

Then the build instruction would be: `docker build -t ardupilot --build-arg UID=$(id -u) --build-arg GID=$(id -g) .` to copy the uid/gid of the local user to the built image. 

The limitation is that the ardupilot image would be useful only on that machine.

khancyr, 2022-05-26:
That is just a generic env builder so we can go use the default UID/GUD. 
Can you make a PR ? that would be a great addition !

zacroker, 2022-05-26:
PR created for this repo - #20854

zacroker, 2022-05-26:
I have also amended the documentation - https://github.com/ArduPilot/ardupilot_wiki/pull/4366

zacroker, 2022-05-27:
New PR - #20857

closed 2022-07-13 by peterbarker