ArduPilot/ardupilot #20859: GCS_Mavlink.cpp passes uninitialised "Vector3f accel_vector" to "copter.mode_guided.set_velaccel()"

ardugitter, 2022-05-27:
## Bug report

**Issue details**

In investigating why the SET_POSITION_TARGET_GLOBAL_INT Mavlink message shows poor responsiveness in Arducopter, I followed the code and discovered that in GCS_Mavlink.cpp in "case MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT", if you request a position target that causes these variables to be set as follows:
pos_ignore == true
**vel_ignore == false**
acc_ignore == true
you end up with function copter.mode_guided.set_velaccel being called which has the uninitialised "Vector3f accel_vector" passed in as an argument. We find that eventually the uninitialised acceleration vector is passed to function copter.Log_Write_Guided_Position_Target().

To avoid encountering this undefined behaviour, you would have to use the appropriate typemask to set both acceleration and velocity in the MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT Mavlink message.

**Version**
4.2.0

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Pixhawk

**Logs**
Please see bug report above. Thank you.

rmackay9, 2022-05-28:
@ardugitter,

Txs very much for the report.

Vector3f are one of the special cases where AP always initailises the internal members ([see here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Math/vector3.h#L79)) so I think it is not quite correct to say an unitialised vector is passed in as an argument.  Instead it's more accurate to say an "all zero" vector3f is passed in.

I think an all zero accel target vector is OK and the position controller will handle it properly.  Can you give more details (and perhaps an onboard log) of the issues you're seeing?  By the way, support issues should really be raised in the forums first but it's no big deal especially considering how developer focussed this issue is.

By the way, your report led me to find this other bug (https://github.com/ArduPilot/ardupilot/issues/20865) so thanks!

ardugitter, 2022-05-28:
Ah, my bad! I had trouble finding the definition of the Vector3f class. Should have looked at the filenames. I'm glad as a consolation that another issue was found.

I posted on the forums [here](https://discuss.ardupilot.org/t/setting-velocity-using-set-position-target-global-int-gives-sluggish-performance/86041) and the params and log are in that post. Perhaps it's best to keep things together in that forum post or would you prefer me to post here?

BTW, I need "agility" and "responsiveness" because of the autonomous drone application I am working on for a startup company. Many thanks!

rmackay9, 2022-05-30:
OK great.  I'll go ahead close this issue then and discussion can continue on the support forum topic. txs

closed 2022-05-30 by rmackay9