ArduPilot/ardupilot #20869: Copter/SITL can't takeoff in guided mode

xlla, 2022-05-30:
## Bug report

**Issue details**

I have followed guide of SITL to test my environment. It launched fine, while try arm and takeoff, it shows "floating point exception - aborting" in parm windows, the link down in console windows.

**Version**
master branch b60d6348ed5246c508c847e7ab0a95d8ce8ac2c6

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
copter

**Hardware type**
SITL

**Logs**
after launch
console
[image]

map
[image]

parm
[image]

cmd
[image]

cmd try takeoff
[image]

console lost link
[image]

parm raise exception
[image]

<details>
<summary>dumpstack.sh_nknown.20259.out</summary>
<pre>
[New Thread 0x1803 of process 20259]
[New Thread 0x1903 of process 20259]
[New Thread 0x2703 of process 20259]
Error calling thread_get_state for GP registers for thread 0x1803

warning: Mach error at "../../gdb/i386-darwin-nat.c:132" in function "virtual void i386_darwin_nat_target::fetch_registers(struct regcache *, int)": (os/kern) invalid argument (0x4)

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/darwin_vers.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/dirstat.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/dirstat_collection.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/init.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/variant.o': can't open to read symbols: No such file or directory.
0x00007fff55961242 in ?? () from /usr/lib/system/libsystem_kernel.dylib
#0  0x00007fff55961242 in ?? () from /usr/lib/system/libsystem_kernel.dylib
No symbol table info available.
#1  0x00007fff558db9de in system () from /usr/lib/system/libsystem_c.dylib
No symbol table info available.
#2  0x000000010b7739b1 in AP_HAL::run_command_on_ownpid(char const*) ()
No symbol table info available.
#3  0x000000010b773e69 in _sig_fpe(int) ()
No symbol table info available.
Backtrace stopped: Cannot access memory at address 0x4f99

Thread 3 (Thread 0x2703 of process 20259):
#0  0x00007fff55960d82 in ?? () from /usr/lib/system/libsystem_kernel.dylib
No symbol table info available.
#1  0x00007fff558db724 in nanosleep () from /usr/lib/system/libsystem_c.dylib
No symbol table info available.
#2  0x00007fff558db618 in usleep () from /usr/lib/system/libsystem_c.dylib
No symbol table info available.
#3  0x000000010b76ff80 in HALSITL::SITL_State::wait_clock(unsigned long long) ()
No symbol table info available.
#4  0x000000010b77249c in HALSITL::Scheduler::delay_microseconds(unsigned short) ()
No symbol table info available.
#5  0x000000010b6ce0b3 in AP_Logger::io_thread() ()
No symbol table info available.
#6  0x000000010b6ceb59 in void Functor<void>::method_wrapper<AP_Logger, &AP_Logger::io_thread>(void*) ()
No symbol table info available.
#7  0x000000010b7729c3 in HALSITL::Scheduler::thread_create_trampoline(void*) ()
No symbol table info available.
#8  0x00007fff55b28661 in _pthread_body () from /usr/lib/system/libsystem_pthread.dylib
No symbol table info available.
#9  0x00007fff55b2850d in _pthread_start () from /usr/lib/system/libsystem_pthread.dylib
No symbol table info available.
#10 0x00007fff55b27bf9 in thread_start () from /usr/lib/system/libsystem_pthread.dylib
No symbol table info available.
#11 0x0000000000000000 in ?? ()
No symbol table info available.

Thread 2 (Thread 0x1903 of process 20259):
#0  0x00007fff55960d82 in ?? () from /usr/lib/system/libsystem_kernel.dylib
No symbol table info available.
#1  0x00007fff558db724 in nanosleep () from /usr/lib/system/libsystem_c.dylib
No symbol table info available.
#2  0x00007fff558db618 in usleep () from /usr/lib/system/libsystem_c.dylib
No symbol table info available.
#3  0x000000010b76ff80 in HALSITL::SITL_State::wait_clock(unsigned long long) ()
No symbol table info available.
#4  0x000000010b77249c in HALSITL::Scheduler::delay_microseconds(unsigned short) ()
No symbol table info available.
#5  0x000000010b772513 in HALSITL::Scheduler::delay(unsigned short) ()
No symbol table info available.
#6  0x000000010b6d4a13 in GCS_MAVLINK::ftp_worker() ()
No symbol table info available.
#7  0x000000010b6d5e49 in void Functor<void>::method_wrapper<GCS_MAVLINK, &GCS_MAVLINK::ftp_worker>(void*) ()
No symbol table info available.
#8  0x000000010b7729c3 in HALSITL::Scheduler::thread_create_trampoline(void*) ()
No symbol table info available.
#9  0x00007fff55b28661 in _pthread_body () from /usr/lib/system/libsystem_pthread.dylib
No symbol table info available.
#10 0x00007fff55b2850d in _pthread_start () from /usr/lib/system/libsystem_pthread.dylib
No symbol table info available.
#11 0x00007fff55b27bf9 in thread_start () from /usr/lib/system/libsystem_pthread.dylib
No symbol table info available.
#12 0x0000000000000000 in ?? ()
No symbol table info available.

Thread 1 (Thread 0x1803 of process 20259):
#0  0x00007fff55961242 in ?? () from /usr/lib/system/libsystem_kernel.dylib
No symbol table info available.
#1  0x00007fff558db9de in system () from /usr/lib/system/libsystem_c.dylib
No symbol table info available.
#2  0x000000010b7739b1 in AP_HAL::run_command_on_ownpid(char const*) ()
No symbol table info available.
#3  0x000000010b773e69 in _sig_fpe(int) ()
No symbol table info available.
Backtrace stopped: Cannot access memory at address 0x4f99
A debugging session is active.

	Inferior 1 [process 20259] will be detached.

Quit anyway? (y or n) [answered Y; input not from terminal]

</pre>
</details>

rmackay9, 2022-05-30:
I'm afraid that I don't have an answer as to what the issue is but I just tested on master under Windows WSL (same as ubuntu) and it seemed to run fine.

Some random things that might help:

- ../waf distclean
- ./waf clean
- git submodule update --recursive
- rm eeprom.bin

khancyr, 2022-05-30:
probably  the same as https://github.com/ArduPilot/ardupilot/issues/19588#issuecomment-1119346114

xlla, 2022-05-30:
hi @rmackay9 , did you known how to trace down the actual reason for that,  build with debug symbols and launch by gdb?

timtuxworth, 2022-05-30:
> did you known how to trace down the actual reason for that, build with debug symbols and launch by gdb?

sim_vehicle.py has  --debug --gdb command line options 
or if you are running on a mac use --lldb instead of --gdb

xlla, 2022-05-31:
@timtuxworth ,  thanks, I use that command to try catch the actual reason, but it works smoothly, the exception was gone!

<pre>
SIM_VEHICLE: Using defaults from (../Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Run ArduCopter (lldb)
SIM_VEHICLE: "/Users/xlla/git/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter (lldb)" "lldb" "-s" "/var/folders/6j/gqhq_r1n0wg5qjypvtkccnqr0000gn/T/tmpv_p769ya" "--" "/Users/xlla/git/ardupilot/build/sitl/bin/arducopter" "-S" "--model" "+" "--speedup" "1" "--slave" "0" "--defaults" "../Tools/autotest/default_params/copter.parm" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--map" "--console"
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Detected vehicle 1:1 on link 0
STABILIZE> Received 1283 parameters (ftp)
Saved 1283 parameters to mav.parm
mode guided
STABILIZE> arm throttle
GUIDED> GUIDED> takeoff 100
GUIDED> Take Off started
Guided (-35.36141498995252, 149.16350253714117) 100.0
Guided (-35.36031912745187, 149.16493754605904) 30.0
arm rtl
GUIDED> usage: arm <check|uncheck|list|throttle|safetyon|safetyoff|safetystatus|bits|prearms>
mode rtl
GUIDED> disarm
RTL> RTL> SIM_VEHICLE: Keyboard Interrupt received ...
SIM_VEHICLE: Killing tasks
</pre>

so I thought it was fine due to firmware rebuild and I have do clean as @rmackay9 suggested, 
<pre>
slim:ArduCopter xlla$ rm eeprom.bin
slim:ArduCopter xlla$ cd ..
slim:ardupilot xlla$ git submodule update --recursive
slim:ardupilot xlla$ cd ArduCopter/
slim:ArduCopter xlla$ ../waf distclean
'distclean' finished successfully (0.543s)
slim:ArduCopter xlla$ ../waf clean
The project was not configured: run "waf configure" first!
</pre>

then switch to normal launch,   

`slim:ArduCopter xlla$ ../Tools/autotest/sim_vehicle.py --map --console`

the exception appears again,  it does even crash the parm window!

parm window log
<pre>
-------------- begin dumpcore.sh output ----------------
[New Thread 0x2603 of process 46664]
[New Thread 0x2703 of process 46664]
[New Thread 0x2803 of process 46664]
Error calling thread_get_state for GP registers for thread 0x2603

warning: Mach error at "../../gdb/i386-darwin-nat.c:132" in function "virtual void i386_darwin_nat_target::fetch_registers(struct regcache *, int)": (os/kern) invalid argument (0x4)

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/darwin_vers.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/dirstat.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/dirstat_collection.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/init.o': can't open to read symbols: No such file or directory.

warning: `/BuildRoot/Library/Caches/com.apple.xbs/Binaries/Libc_darwin/install/TempContent/Objects/Libc.build/libsystem_darwin.dylib.build/Objects-normal/x86_64/variant.o': can't open to read symbols: No such file or directory.
0x00007fff55960d82 in ?? () from /usr/lib/system/libsystem_kernel.dylib
/tmp/gdb.46703:2: Error in sourced command file:
Can't create a corefile
-------------- end dumpcore.sh output ----------------
Abort trap: 6
slim:ArduCopter xlla$ 
</pre>
 
cmd window
<pre>
[882/882] Linking build/sitl/bin/arducopter
Waf: Leaving directory `/Users/xlla/git/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /Users/xlla/git/ardupilot/build/sitl
Target          Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
---------------------------------------------------------------------------------
bin/arducopter   2416640    356352        0               2772992  Not Applicable

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (2m38.804s)
SIM_VEHICLE: Using defaults from (../Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/Users/xlla/git/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/Users/xlla/git/ardupilot/build/sitl/bin/arducopter" "-S" "--model" "+" "--speedup" "1" "--slave" "0" "--defaults" "../Tools/autotest/default_params/copter.parm" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--map" "--console"
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Detected vehicle 1:1 on link 0
STABILIZE> Received 1283 parameters (ftp)
Saved 1283 parameters to mav.parm
mode guided
STABILIZE> arm throttle
GUIDED> GUIDED> takeoff 100
GUIDED> Take Off started
EOF on TCP socket
Attempting reconnect
[Errno 61] Connection refused sleeping
[Errno 61] Connection refused sleeping
Attempting reconnect

</pre>

However I can use it at debug mode, I'm glad to know it.

shiv-tyagi, 2022-06-13:
Hi @xlla! The issue you are facing is a known issue among few developers using clang on macos to compile the binaries. I faced the similar issue and I fixed it by installing gcc/g++ from homebrew and compiling the binaries using the same.
This [comment](https://discuss.ardupilot.org/t/sitl-error-on-macos-monterey-12-1-almost-solved/79996/21?u=shiv-tyagi) on a related blog post might help.

xlla, 2022-06-13:
@shiv-tyagi , wow! after change compiler to gcc-11, it works fine in release mode. thank you!
it seems the clang does not compatible with ArduPilot.

closed 2022-06-13 by xlla