ArduPilot/ardupilot #20875: sim_vehicle apparently unable to handle models with >65.54v

AndKe, 2022-05-31:
**Issue details**
sim_vehicle battery voltage seems to be unintentionally limited to 65.53 volts.  
param set SIM_BATT_VOLTAGE 90   sets the voltage to 65.53v  (65535mV ?)  which looks like the limit of an unsigned short. 

Please increase the maximum allowed voltage.

**Version**
current master

**Platform**
[ ] All
[  ] AntennaTracker
[x ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X8

**Hardware type**
n/a

**Logs**

timtuxworth, 2022-05-31:
This is likely related to [https://github.com/ArduPilot/ardupilot/issues/19588](url), in that it seems to be a similar issue of values being passed as unsigned ints. I'm guessing without seeing the code.

AndKe, 2022-06-01:
@timtuxworth  it seems like you are onto something - do you have any idea if this requires only a mall fix - and can be corrected easily/soon?

timtuxworth, 2022-06-01:
> @timtuxworth it seems like you are onto something - do you have any idea if this requires only a mall fix - and can be corrected easily/soon?

I don't know the code very well, but it looks to me like its baked into the CAN protocol, so I'm thinking it might not be easy to fix.

davidbuzz, 2022-06-01:
an inmmediately available work-around is to change your calculated ratios so they report 1/10th the real voltage, its still pretty easy to read on the gcs, its just the decimal point in the wrong place.