ArduPilot/ardupilot #20880: Rover: Encoder bad timestamp

MagicHardware, 2022-06-01:
**Issue details**
https://discuss.ardupilot.org/t/wheel-rate-control-tuning-problem/85863

Ther is a bad timestamp in the encoder code. It should be converted to ms, now it is us. The error affects WRC and EKF. There get wrong rate of wheels. I checked the fix and it seems to work well.

In code:
https://github.com/ArduPilot/ardupilot/blob/e71ced798486bbee47247e1f2c8a15b7b6a6e11f/libraries/AP_WheelEncoder/WheelEncoder_Quadrature.cpp#L141

Should be:
`irq_state.last_reading_ms = timestamp * 1e-3f;`

**Version**
4.1,4.2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
Tested on : skid steer rover

**Hardware type**
Tested on: CUAV V5 Plus

**Logs**

rmackay9, 2022-06-01:
@MagicHardware,

Oh my, yes indeed that looks like a bug.  Thanks very much for finding this.  The timestamp argument is very badly named.  It's a uint32_t so normally that would mean milliseconds.. but it's micros.  thanks again.

MagicHardware, 2022-06-01:
No problem 😄

IamPete1, 2022-06-01:
A PR would be great, 

https://ardupilot.org/dev/docs/submitting-patches-back-to-master.html

rmackay9, 2022-06-07:
This is fixed in master now so closing.  Thanks @MagicHardware!

closed 2022-06-07 by rmackay9