ArduPilot/ardupilot #20883: First mission objective is being skipped in SITL and MAVROS.

eMrazSVK, 2022-06-01:
## Bug report

Hello, I am implementing API. Let's say it is similar to DroneKit, but it works on top of Mavros. I know it might be Mavros related, but [the error](https://github.com/ArduPilot/ardupilot/blob/Copter-4.2/ArduCopter/mode_auto.cpp#L31) I get is clearly raised by ArduPilot. 

The problem is, that I push waypoints (mission commands) to the ArduPilot (SITL) via Mavros interface (`/mavros/mission/push` service). Then I can read those mission items from topic `/mavros/mission/waypoints` and it looks GOOD:
```
current_seq: 0
waypoints: 
  - 
    frame: 3
    command: 22
    is_current: True
    autocontinue: True
    param1: 0.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 10.0
  - 
    frame: 3
    command: 19
    is_current: False
    autocontinue: True
    param1: 5.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 0.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 1.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 13.0
  - 
    frame: 3
    command: 19
    is_current: False
    autocontinue: True
    param1: 5.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 0.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 1.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 15.0
  - 
    frame: 3
    command: 21
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 0.0
---
```
Now what is strange - and also the reason why I think this is an ArduPilot issue is the fact, that `wp list` alters uploaded waypoints. If I use `wp list` command inside SITL console, the ouput looks like this. The first command is suddenly changed to Waypoint. So this way, if I try to start the mission, I get error mentioned before `Auto: Missing Takeoff Cmd` :
```
wp list
GUIDED> Requesting 6 waypoints t=Wed Jun  1 13:09:29 2022 now=Wed Jun  1 13:09:29 2022
16 0 -35.3632622000 149.1652375000 584.070007 p1=0.0 p2=0.0 p3=0.0 p4=0.0 cur=0 auto=1
19 3 0.0000000000 0.0000000000 0.000000 p1=5.0 p2=0.0 p3=1.0 p4=0.0 cur=0 auto=1
16 3 0.0000000000 0.0000000000 13.000000 p1=1.0 p2=0.0 p3=0.0 p4=0.0 cur=0 auto=1
19 3 0.0000000000 0.0000000000 0.000000 p1=5.0 p2=0.0 p3=1.0 p4=0.0 cur=0 auto=1
16 3 0.0000000000 0.0000000000 15.000000 p1=1.0 p2=0.0 p3=0.0 p4=0.0 cur=0 auto=1
21 3 0.0000000000 0.0000000000 0.000000 p1=0.0 p2=0.0 p3=0.0 p4=1.0 cur=0 auto=1
```
I can also check waypoints ROS topic once again and the first waypoint is changed there too:
```
urrent_seq: 0
waypoints: 
  - 
    frame: 0
    command: 16
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: -35.3632622
    y_long: 149.1652375
    z_alt: 584.070007324
  - 
    frame: 3
    command: 19
    is_current: False
    autocontinue: True
    param1: 5.0
    param2: 0.0
    param3: 1.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 0.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 1.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 13.0
  - 
    frame: 3
    command: 19
    is_current: False
    autocontinue: True
    param1: 5.0
    param2: 0.0
    param3: 1.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 0.0
  - 
    frame: 3
    command: 16
    is_current: False
    autocontinue: True
    param1: 1.0
    param2: 0.0
    param3: 0.0
    param4: 0.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 15.0
  - 
    frame: 3
    command: 21
    is_current: False
    autocontinue: True
    param1: 0.0
    param2: 0.0
    param3: 0.0
    param4: 1.0
    x_lat: 0.0
    y_long: 0.0
    z_alt: 0.0
---
```
It happens with anything I put at the start of the mission. I tried taking off in `guided` mode and switching to AUTO mode after the takeoff height is reached, but whatever mission item I put at front of the mission, it just gets replaced. 

**Version**
Copter 4.2.0

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad/Plus

**Hardware type**
SITL

chcaya, 2025-02-07:
Hello,

I’m facing the same issue, and even after using the [/mavros/cmd/set_home](https://docs.ros.org/en/hydro/api/mavros/html/srv/CommandHome.html) service before pushing my waypoints, I still receive the same error message` Auto: Missing Takeoff Cmd`. I modified the [AUTO_OPTIONS](https://ardupilot.org/copter/docs/parameters.html#auto-options) parameters accordingly so that if I upload the mission with QGC, the drone takes off and completes the mission without any issue. I can bypass the error message ` Auto: Missing Takeoff Cmd` by sending two `NAV_TAKEOFF` commands as the first waypoints, but the drone still doesn't take off in `AUTO` mode. I have to take off in `GUIDED` mode, and then switch to `AUTO` mode to complete the mission.

To be clear, I have no error messages coming from ROS, Mavlink console or QGC, and the drone is properly armed in `AUTO` mode with its GPS fix. Also, I am using ArdupilotSITL and Gazebo.

I feel like these are some sketchy workarounds, are there any better ways to get the expected behavior in `AUTO` mode without having to send two `NAV_TAKEOFF` commands?

If it may help, here is my waypoint list:

```
waypoint_msg = Waypoint()
waypoint_msg.frame = Waypoint.FRAME_GLOBAL_REL_ALT
waypoint_msg.command = CommandCode.NAV_TAKEOFF
waypoint_msg.is_current = False
waypoint_msg.autocontinue = True
waypoint_msg.param1 = 0.0
waypoint_msg.param2 = 0.0
waypoint_msg.param3 = 0.0
waypoint_msg.param4 = 0.0
waypoint_msg.x_lat = self.home_position_msg.geo.latitude
waypoint_msg.y_long = self.home_position_msg.geo.longitude
waypoint_msg.z_alt = 50.0
self.waypoint_push_req.waypoints.append(waypoint_msg)

waypoint_msg = Waypoint()
waypoint_msg.frame = Waypoint.FRAME_GLOBAL_REL_ALT
waypoint_msg.command = CommandCode.NAV_TAKEOFF
waypoint_msg.is_current = False
waypoint_msg.autocontinue = True
waypoint_msg.param1 = 0.0
waypoint_msg.param2 = 0.0
waypoint_msg.param3 = 0.0
waypoint_msg.param4 = 0.0
waypoint_msg.x_lat = self.home_position_msg.geo.latitude
waypoint_msg.y_long = self.home_position_msg.geo.longitude
waypoint_msg.z_alt = 50.0
self.waypoint_push_req.waypoints.append(waypoint_msg)

waypoint_msg = Waypoint()
waypoint_msg.frame = Waypoint.FRAME_GLOBAL_REL_ALT
waypoint_msg.command = CommandCode.NAV_WAYPOINT
waypoint_msg.is_current = False
waypoint_msg.autocontinue = True
waypoint_msg.param1 = 0.0
waypoint_msg.param2 = 0.5
waypoint_msg.param3 = 0.0
waypoint_msg.param4 = 0.0
waypoint_msg.x_lat = self.home_position_msg.geo.latitude-0.0001
waypoint_msg.y_long = self.home_position_msg.geo.longitude-0.0001
waypoint_msg.z_alt = 50.0
self.waypoint_push_req.waypoints.append(waypoint_msg)

waypoint_msg = Waypoint()
waypoint_msg.frame = Waypoint.FRAME_GLOBAL_REL_ALT
waypoint_msg.command = CommandCode.NAV_WAYPOINT
waypoint_msg.is_current = False
waypoint_msg.autocontinue = True
waypoint_msg.param1 = 0.0
waypoint_msg.param2 = 0.5
waypoint_msg.param3 = 0.0
waypoint_msg.param4 = 0.0
waypoint_msg.x_lat = self.home_position_msg.geo.latitude+0.0001
waypoint_msg.y_long = self.home_position_msg.geo.longitude+0.0001
waypoint_msg.z_alt = 50.0
self.waypoint_push_req.waypoints.append(waypoint_msg)

waypoint_msg = Waypoint()
waypoint_msg.frame = Waypoint.FRAME_GLOBAL_REL_ALT
waypoint_msg.command = CommandCode.NAV_RETURN_TO_LAUNCH
waypoint_msg.is_current = False
waypoint_msg.autocontinue = True
waypoint_msg.param1 = 0.0
waypoint_msg.param2 = 0.0
waypoint_msg.param3 = 0.0
waypoint_msg.param4 = 0.0
waypoint_msg.x_lat = 0.0
waypoint_msg.y_long = 0.0
waypoint_msg.z_alt = 0.0
self.waypoint_push_req.waypoints.append(waypoint_msg)
```

Thank you

aksha787, 2025-07-03:
@chcaya did you fix the issue ? i am facing the same issue

chcaya, 2025-07-03:
> [@chcaya](https://github.com/chcaya) did you fix the issue ? i am facing the same issue

@aksha787 Not really, I simply switched from using `AUTO` mode to `GUIDED` mode for the whole mission while operating with my companion computer.