ArduPilot/ardupilot #20884: FFT_ENABLE,1 with ARMING_REQUIRE,0 causes watchgot reset

Jaaaky, 2022-06-01:
## Bug report

**Issue details**

I found a bug related to GyroFFT, .. while testing.. I set  ARMING_REQUIRE,0  when FFT_ENABLE,1 .. Actually it cause a watchdog reset every single time.
I reset everything to default, only changed FFT_ENABLE,1 and ARMING_REQUIRE,0 it's enough to trigger watchdog reset every single time at boot..

I tried to bisect but seems the issue is very old, went back to Jun 2020 commit `9013432d2715de4611649fa90b33fc9f073244c9` and it's still there..

**Version**
Using arduplane master on fmuv3 board 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Fmuv3

andyp1per, 2022-06-04:
I can't reproduce this. I think it's user error. Tried on copter and plane on a Durandal - all worked fine.

@Jaaaky note that an easy mistake to make is that if your accels are not calibrated the FFT won't work. This tripped me up a couple of times.

Jaaaky, 2022-06-05:
The issue is reproducible for sure.. I've just reproduced it again on ArduPlane-stable 4.2.1 on pixhawk1

Accels are calibrated and I even get `AP: FFT: calibrated 1.0KHz/7.8Hz/93.8Hz` on startup .. Then just set `ARMING_REQUIRE,0` causes failure on every startup.
It's not random, it fails every single time shortly after boot.

Jaaaky, 2022-06-06:
I could just reproduce the `ARMING_REQUIRE,0` on sitl on ArduPlane-stable branch .. It crashes every single at boot once activated.. 
Only other changed params are:
```
FFT_ENABLE,1
INS_HNTCH_ENABLE,1
RPM1_TYPE,4
```

tridge, 2022-06-06:
@andyp1per tells me he has found the issue with ARMING_REQUIRE=0, and we discussed a fix

closed 2022-06-14 by tridge