ArduPilot/ardupilot #20886: SERIALx_OPTIONS "don't forward" breaks MAVLink gimbals

rmackay9, 2022-06-02:
While working on improving our MAVLink gimbal support (in particular the Gremsy gimbal) I've found that setting SERIALx_OPTIONS = 1024 completely breaks the ability to use mavlink gimbals.  It seems that messages are not being passed between the autopilot and the gimbal once this option is turned on.

This is quite annoying because there are tons of messages sent between the gimbal and autopilot.  Not being able to stop the messages from the gimbal being forwarded to the GCS means the GCS<->autopilot bandwidth is completely consumed by the gimbal mavlink traffic.

Related issue https://github.com/ArduPilot/ardupilot/issues/20723

muramura, 2022-06-05:
Is it MOUNT_ORIENTATION that is using a lot of bandwidth?
Can you identify the version of the gimbal that is causing this?

GIMBAL SETUP:
https://support.gremsy.com/support/solutions/articles/36000189926-how-to-setup-gremsy-gimbal-with-pixhawk-cube-

rmackay9, 2022-06-16:
This is fixed in master now so closing.

@muramura, sorry I didn't see your report.  The [setup instructions for the new gremsy driver are here](https://discuss.ardupilot.org/t/ardupilot-4-3-gremsy-gimbal-improvements-testers-wanted/86479).  I'm not sure exactly which messages from the gimbal consume the most bandwidth but I suspect it is the attitude reporting messages.

closed 2022-06-16 by rmackay9