ArduPilot/ardupilot #20887: It is possible to create two drivers for the same BARO

Williangalvani, 2022-06-02:
The way AP_Baro.cpp is currently structured, a driver can be instantiated at init() and another one after probing if `BARO_EXT_BUS` and `BARO_PROBE_EXT` are set.

In my case, I had it probed [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro.cpp#L713-L714) and [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro.cpp#L883-L886)

[image]
[image]

Tested using a Navigator (linux) board with ardusub master.

While this happened in a sub-specific block, it looks like it could happen with other boards mentioned in `init()`, too.

Could we maybe check if the devid is already registered?

ES-Alexander, 2022-06-02:
A workaround for [the Sub-specific issue](https://discuss.bluerobotics.com/t/pixhawk-4-i2c-sensors-not-found/11507/7) is to un-set `ms5837` in `BARO_PROBE_EXT` since it's currently checked for automatically. That's unideal because it actively encourages unintuitive/incorrect behaviour, but good to at least have some way of getting the intended functionality working in the meantime.

HefnySco, 2022-09-01:
@Williangalvani this issue should be fixed by [PR #20958](https://github.com/ArduPilot/ardupilot/pull/20958)

closed 2022-09-01 by HefnySco