ArduPilot/ardupilot #20898: RPM Harmonic notch based on FFT not working on master

Jaaaky, 2022-06-03:
## Bug report

**Issue details**

Testing RPM sensor of Harmonic notch based on FFT it's working fine on ArduPlane-stable - 4.2.1 - but not on master..
I simply test rpm of a ICE piston engine.
On master it displays a constant RPM on 4800 

Bisecting the issue I found it was caused by pr #20521 
Here is the git bisect result:

The first bad commit could be any of:
1cd1519a4f770218567642c3cc5d66c56bb41659
7aafd5cf9869884efa353915ae180b9abc0878bb
d57abb1230e88e263737e69acdb9b46485be3587
13107dfa6dee4589bbfa8ff2df6891d39b71fd2e
ae1e9e06c184ec3916fdaf78ef04228fa1f12b32
7272a29b7b713f6d7ca48d296d996baadab16730
782e4887bc093cb27c7f67147614c2cc28462e6e
c03436b5e4802bd54e420b79d09d5a853fcbeb48
d7d04bc7cfdee6984f1e01657f90428a5f4f9d75
c1d841dd934f3d4151edbb85df0201b0d9f02fd2
e2e1e74da5e570a4fcc65fbf68e2b0a482c2cedb
8d187f4bc64e1121630de566fc71a5e6095e2e9a
ddab3f02ca13732cc8849e8cbffa066b7ca12ac7
86db91e3b456aa7abebddbdd7b8082311fd830d3
a2fc4d909d87efe739039fdab8c094906d6ee94e
ac50d3cc9ff8a84c99a37b256716de659cc280ab
e6b56ce9baa56149b72053a4c9bf41c5af634e7a
e67d9b41795690fde7ddaa287259586815d5be53
eb4a70f6134f91529d9adec0d362e6a580e1819a
15084cb6f3079880e0be95c9f72ff7d7490e8a9d
622ad2a088edb689416d95949cd04dbdfc2d22eb
301c56d30a7b31a607d4af2a426193a20932d217
We cannot bisect more!

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Plane

**Hardware type**
Tested on 4 different boards.. from fmuv3 to fmuv5x

**Logs**
I've tried different params/settings, here are some logs on master code
[LOG_fail_master.zip](https://github.com/ArduPilot/ardupilot/files/8831701/LOG_fail_master.zip)

Jaaaky, 2022-09-12:
@tridge @peterbarker 
The bug is now on ArduPlane stable 4.2.3 too!

Jaaaky, 2022-09-12:
Git bisect results on ArduPlane-stable:

**There are only 'skip'ped commits left to test.
The first bad commit could be any of:**
a23a25f6b30e17f8d7fb85152c695c534ffb69ad
b429bb7b989c1b94abc79602ac6d892c8573204e
0467c1aef6e4a0393920c9615bf1adb51e9b8489
5e1489eeff93997fe8e1dd3c2bddb77faab464b6
711ec6ce70d83ff4101d8d004f635666caeb8f2c
e763386651114aaf27bc5fbd4f8de366c7943925
1de23e564eaf9c53f461663023fec45bc7017914
559f4c00c3121644be5729e987b317fe78c0f176
a617057f36623fff781e5fd55f97b02019ad1f95
e799649e6c12a4aee048c1d33ea8daae2729fd62
d893d9bc89d1ab98298b7fad06d8c0a95543c9e3
e894dfe3391fee01c91b4d7b54a38f2ca4e8f977
4c5ea5dfdf6457d348b36e784eb128eb72ab8cf4
5080c32197fab61605c1921b472a53dd4e53487c
f8c1c398684adb1ab278f1d9383470ac851ef3fb
f9de6dff1cee7d0809c261c425ee2486566e602f
2a5d662d8479829f646c61a243e269f02aaf5fec
3d025a7c5c4f6ca5cd29983a6f6037711605ea30
25d00e433e6b2b2051d501c71e0fe40daac89891
5f2cc8960d3182eb93653539ab3bf6fa08f5e447
975fc6b70de27d64270786f768b047a741cc0e86
**We cannot bisect more!**

rmackay9, 2022-09-13:
@andyp1per, maybe you could help us investigate this?

andyp1per, 2022-09-13:
@Jaaaky can you post your parameters please and tell me which board you are using. Can you also verify that this is still broken for you on master as there have been a number of bug fixes and changes there

Jaaaky, 2022-09-13:
@andyp1per 
Tested on multiple boards, recently on Pixhawk1 same parameters mentioned in issue #20898

```
FFT_ENABLE,1
FFT_MINHZ,20
FFT_MAXHZ,400
INS_HNTCH_ENABLE,1
INS_HNTCH_MODE,4
INS_HNTCH_REF,1
FFT_WINDOW_SIZE,64
RPM1_TYPE,4
```

Jaaaky, 2022-09-13:
@andyp1per Yes I can confirm it's broken both on master and ArduPlane-stable 

Related issue
#20908

andyp1per, 2022-09-13:
@Jaaaky is this on plane?

Jaaaky, 2022-09-13:
@andyp1per 
Yes, it's already mentioned in main bug post.

andyp1per, 2022-09-13:
Fix here: https://github.com/ArduPilot/ardupilot/pull/21717

rmackay9, 2022-09-13:
@Jaaaky, thanks very much for the report!

closed 2022-09-14 by tridge