ArduPilot/ardupilot #20908: FFT is not started/calibrated if LOG_DISARMED,1 on ArduPlane-stable 4.2.1

Jaaaky, 2022-06-05:
## Bug report

**Issue details**

Changing only the following params from default:
```
FFT_ENABLE,1
FFT_MINHZ,20
FFT_MAXHZ,400
INS_HNTCH_ENABLE,1
INS_HNTCH_MODE,4
INS_HNTCH_REF,1
FFT_WINDOW_SIZE,64
RPM1_TYPE,4
```

I get a message like this `AP: FFT: calibrated 1.0KHz/7.8Hz/93.8Hz` on startup and harmonic rpm works fine 

If I enable `LOG_DISARMED,1` FFT is not started, and rpm shows a random constant value.

**Version**
ArduPlane-stable 4.2.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Pixhawk1

Jaaaky, 2022-06-06:
I can confirm the `LOG_DISARMED,1` bug  can be reproduced on Pixhawk5x too.

I've done the following, to make sure nothing else is interfering:

1. Create fft.param file with the following content:
```
FFT_ENABLE,1
FFT_MINHZ,20
FFT_MAXHZ,400
INS_HNTCH_ENABLE,1
INS_HNTCH_MODE,4
INS_HNTCH_REF,1
FFT_WINDOW_SIZE,64
RPM1_TYPE,4
```

2. Build and upload on ArduPlane-stable branch 
`$ ./waf distclean && git submodule update --init --recursive && ./waf configure --default-parameters=fft.param --disable-scripting --board=Pixhawk5X && ./waf -p plane --upload`

3. After flashing, using mavproxy, do ` param set FORMAT_VERSION 0` to reset eeeprom.

Make sure a formatted SDcard is inserted.

4. Calibrate accels and reboot

5. Do a reboot, check for the following message:
`AP: FFT: calibrated 2.0KHz/15.7Hz/200.0Hz`

reboot, multiple time, you'll find similar message at each boot normally

6. Do `param set LOG_DISARMED 1` then reboot.. fft is not calibrated on startup any more. You can try multiple reboots too.

7. Again `param set LOG_DISARMED 0` then reboot, fft is calibrated normally again.

Jaaaky, 2022-06-06:
I actually found a more serious problem Pixhawk5X too. The reboot command after activating `LOG_DISARMED,1` following the mentioned steps, does not work fine, I've to do a power cycle..

andyp1per, 2022-06-06:
@Jaaaky I still cannot reproduce this in my environment following your instructions - it all works fine, I definitely think it is memory/RAM related

Jaaaky, 2022-06-06:
@andyp1per Have you seen my update on Pixhawk5x it has free mem of over 180KBs at least ?

Which board you've tested?

I'm sure it's not a ram issue.

tridge, 2022-06-24:
@Jaaaky I reproduced, and the issue is the board was failing another arming check in ins_checks()
if you get it to pass all other arming checks then FFT should init
@andyp1per we really should move FFT init out of arming checks

peterbarker, 2022-09-01:
@Jaaaky did you get past this problem?  Do you know if it still a problem on more recent versions?

Jaaaky, 2022-09-12:
@peterbarker Yes, this issue still exists on Stable branch - currently arduplane 4.2.3 -, it's not on master.

Also, another issue which was reported against master about fft rpm not working, now also exists on ArduPlane-stable 4.2.3 
https://github.com/ArduPilot/ardupilot/issues/20898

Jaaaky, 2022-09-12:
@peterbarker 
Also, with ARMING_REQUIRE,0 FFY is not started/calibrated on boot too, even with LOG_BACKEND_TYPE,0 and 
 accelerometers are calibrated.
 Confirmed on master and stable

Jaaaky, 2022-09-13:
@andyp1per Please check here too. They're all related

andyp1per, 2022-09-13:
@Jaaaky they are all different issues