ArduPilot/ardupilot #20934: BMI270 drivers support for omnibusf4sd

rolfbartels, 2022-06-08:
## Feature request

**Is your feature request related to a problem? Please describe.**
Due to the MCU6000 no longer being produced omnibusf4 controller now have BMI270 and there is no supporting firmware.

**Describe the solution you'd like**
Here is an output from BetaFLight
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.26V, Core temp=38degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4096, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456

I would like a firmware compiled to support the new GYRO and ACC. I am able to assist with testing if required.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**dditional context**
02 Jun 2022 11:57:04 : Config Error: INS: unable to initialise driver
02 Jun 2022 11:57:04 : Config Error: fix problem then reboot

I have an open thread here
https://discuss.ardupilot.org/t/hardware-support-omnibusf4sd/86363

andyp1per, 2022-06-09:
Great! You can follow the process here: https://ardupilot.org/dev/docs/submitting-patches-back-to-master.html

rolfbartels, 2022-06-09:
I could use some help

I have created a folder omnibusf4sd copied from omnibusf4
My hwdef.dat looks like this

`include ../omnibusf4pro/hwdef.dat

undef APJ_BOARD_ID
undef PC8
undef PC9
undef PA4
undef IMU

# board ID for firmware load
APJ_BOARD_ID 1002

#Omnibus F4 AIO V3 and later had hw inverter on UART6
#Overide it to use as GPS UART port
PC8 SBUS_INVERT_RX OUTPUT LOW
PC9 SBUS_INVERT_TX OUTPUT LOW

#Overside for BMI270
PA4 BMI270_CS CS
SPIDEV mpu6000   SPI1 DEVID1 BMI270_CS  MODE3  1*MHZ  4*MHZ
SPIDEV bmi270 SPI1 DEVID1 BMI270_CS MODE3 10*MHZ 10*MHZ
IMU BMI270 SPI:bmi270 ROTATION_ROLL_180_YAW_225
`
I am getting a error when I run the ./waf
Writing hwdef setup in /home/rolf/ardupilot/build/omnibusf4sd/hwdef.h
Error: Bad CS pin in SPIDEV line ['mpu6000', 'SPI1', 'DEVID1', 'MPU6000_CS', 'MODE3', '1*MHZ', '8*MHZ']

The only way I can get the run to complete is if I # out the line 110 in the ../omnibusf4pro/hwdef.dat
`SPIDEV mpu6000    SPI1 DEVID1 MPU6000_CS MODE3 1*MHZ 8*MHZ`

But then I am changing the omnibusf4pro hwdef. 
How do i do a override ? if this is not possialbe then I would need to copy the omnibusf4pro and just edit that as the new omnibusf4sd

andyp1per, 2022-06-09:
You should add the definition direction to the omnibusf4pro hwdef rather than creating a new one.

See how I did it here: https://github.com/ArduPilot/ardupilot/blob/d85fe81537e08a22cdf3d50b37883fad12c34e8f/libraries/AP_HAL_ChibiOS/hwdef/MambaH743v4/hwdef.dat#L159

peterbarker, 2024-07-31:
bmi260 is now available on the omnibusf4pro - closing this one.

closed 2024-07-31 by peterbarker