ArduPilot/ardupilot #20939: Copter: ascending in circle mode ignoring altitude fence limit.

muramura, 2022-06-08:
## Bug report

**Issue details**

In circle mode, when throttle is increased, it rises ignoring the fence.
FENCE action is report only.

**Version**

ArduCopter V4.3.0-dev

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Copter quad

**Hardware type**

SITL

**Logs**
[image]

rmackay9, 2022-06-09:
@muramura,

Great, thanks for finding this.  I'm sure the issue is that we are missing a line like this:

```
        // get avoidance adjusted climb rate
        target_climb_rate = get_avoidance_adjusted_climbrate(target_climb_rate);
```

Do you want to create a PR for this?  No pressure of course..

closed 2022-06-13 by rmackay9

muramura, 2022-06-13:
@shiv-tyagi san, @rmackay9 san. I could confirm the renovation with SITL. Thanks.

PR #20964 (2022-06-13): Copter: avoid ascend beyond fence's max alt in circle mode
Fixes #20939. The climb rate must be such that the vehicle does not breach the fence. 
I gave it a test on sitl and it worked well.
@rmackay9 Would be great if you could have a look once, please.