ArduPilot/ardupilot #20940: Vectored tailsitter: abnormal behavior of tilt motors

robustini, 2022-06-09:
## Bug report
I have ascertained that in vector tailsitters, which can therefore take off lying on the ground in the classic attitude of plane flight, if you manually set an attitude to hit the tilt servos do not remain as they should at 90 degrees (upwards) but they even do move in the direction of flight.
The video speaks more than any other detail.

**Issue details**

https://user-images.githubusercontent.com/3864712/172853954-3144af1b-3c0a-4973-8706-601b86f150d3.mp4

**Version**

The master of today.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
3D Printed tailsitter vectored

**Hardware type**
Cube Orange

robustini, 2022-10-11:
@IamPete1 @tridge @lthall could anyone look at us?

IamPete1, 2022-10-11:
This is the problem code. Because were doing euler angle error pitch goes up to 90 and then gets smaller again.  

https://github.com/ArduPilot/ardupilot/blob/938df192fa5eaccfb97fc340c8eace4e3d41774b/ArduPlane/tailsitter.cpp#L421-L428