ArduPilot/ardupilot #20955: VTX stick menu does not work with HDZERO (and a fix sort of)

pbreed, 2022-06-11:
## Bug report
The VTX stick menu is not accessible using HDZERO and likely not for ANY vtx using MSP
It is not accessible because the code is not correctly processing VTX MSP requests.

**Issue details**
Three errors, all three need fixed for this to work.
1)Error # 1
Iibraries/AP_MSP/AP_MSP_Telem_Backend.cpp
In the 
void AP_MSP_Telem_Backend::process_incoming_data()
function we have code:
            const uint8_t c = _msp_port.uart->read();
This is an error if you look at libraries\AP_HAL_ChibiOS\UartDriver.h you will see that
    int16_t read() override;
This returns a SIGNED value not an unsinged one.
For certain error conditions it returns -1.
The system should not try and parse -1 so I did this:
			int16_t rv= _msp_port.uart->read();
			if(rv>=0){   
				uint8_t c=rv;
				msp_parse_received_data(&_msp_port, c);
		
            if (_msp_port.c_state == MSP_COMMAND_RECEIVED) {
                msp_process_received_command();
            }

2(Error #2 Why is read returning -1?
\AP_HAL_ChibiOS\UartDriver.cpp
In the function :
ssize_t UARTDriver::read(uint8_t *buffer, uint16_t count)
This ( check  is failing:
    if (lock_read_key != 0 || _uart_owner_thd != chThdGetSelfX()){
        return -1;
    }
 Specifically if I comment out the uart_owener_thd check it works.
 Not exactly sure how to fix this correctly....
 
 3)Error #3
 Per the MSP V1 specification the packet starts with 
 $M< for packets going to the master
 $M> for packets  coming from the master //I may have this backwards
 and $M! for errors.

 libraries\AP_MSP\msp.cpp 
 bool MSP::msp_parse_received_data(msp_port_t *msp, uint8_t c)
Only handles headers with <

So I changed :
    case MSP_HEADER_M:      // Waiting for '<'
        if (c == '<') {
 to:
    case MSP_HEADER_M:      // Waiting for '<' or '>'
            if ((c == '<') || (c=='>')){ 

These three changes fixed the issue for me and VTX stick menu now works.
Would like some advice on the proper fix for Item #2 the thread owner issue...  

**Version**

**Platform**
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
MATEK H743 likelt all STM32 based FC

**Logs**
NA

yaapu, 2022-06-17:
Thread ownership issue confirmed

yaapu, 2022-06-17:
Fix proposed here https://github.com/ArduPilot/ardupilot/pull/20997

yaapu, 2022-07-04:
@pbreed can you please check if the fix solves the issue?
thanks

closed 2023-01-09 by yaapu