ArduPilot/ardupilot #20956: Improper interface to UART

pbreed, 2022-06-11:
## Bug report
UART reads are coded incorrectly.
libraries\AP_HAL_ChibiOS\UartDriver.h you will see that
int16_t read() override;
This returns a SIGNED value not an unsinged one.
For certain error conditions it returns -1.
In the following places this result is directly cast as a char.
It will both miss errors, and get in an infinite read loop if its checking for data...
on any error that returns -1.

This is broken in so many places.. that one probably should kill the read function and 
replace it with a readi16 and readu8  or something similar to make sure you catch it every where...

**Issue details**
All of the following files have this issue in one or more places.
This was a simple grep for read() and an examination for the grep line being obviously wrong for a signed value.
Its possible that this is wrong in more places as well.

ardupilot\libraries\AP_Camera\AP_RunCam.cpp
ardupilot\libraries\AP_EFI\AP_EFI_Serial_MS.cpp
ardupilot\libraries\AP_GPS\AP_GPS.cpp:
ardupilot\libraries\AP_GPS\AP_GPS_GSOF.cpp: 
ardupilot\libraries\AP_GPS\AP_GPS_NMEA.cpp: 
ardupilot\libraries\AP_GPS\AP_GPS_NOVA.cpp: 
ardupilot\libraries\AP_GPS\AP_GPS_SBF.cpp:  
ardupilot\libraries\AP_GPS\AP_GPS_SBP.cpp:  
ardupilot\libraries\AP_GPS\AP_GPS_SBP2.cpp: 
ardupilot\libraries\AP_Hott_Telem\AP_Hott_Telem.cpp
ardupilot\libraries\AP_IOMCU\AP_IOMCU.cpp
ardupilot\libraries\AP_Proximity\AP_Proximity_Cygbot_D1.cpp
ardupilot\libraries\AP_Proximity\AP_Proximity_RPLidarA2.cpp
ardupilot\libraries\AP_Proximity\AP_Proximity_TeraRangerTower.cpp
ardupilot\libraries\AP_Proximity\AP_Proximity_TeraRangerTowerEvo.cpp
ardupilot\libraries\AP_RangeFinder\AP_RangeFinder_LightWareSerial.cpp:
ardupilot\libraries\AP_RangeFinder\AP_RangeFinder_MaxsonarSerialLV.cpp
ardupilot\libraries\AP_RangeFinder\AP_RangeFinder_NMEA.cpp
ardupilot\libraries\AP_RangeFinder\AP_RangeFinder_USD1_Serial.cpp
ardupilot\libraries\AP_RangeFinder\AP_RangeFinder_Wasp.cpp:      
ardupilot\libraries\AP_VideoTX\AP_SmartAudio.cpp
ardupilot\libraries\AP_WindVane\AP_WindVane_NMEA.cpp:
ardupilot\libraries\GCS_MAVLink\GCS_Common.cpp:      
ardupilot\Tools\AP_Periph\adsb.cpp:      
ardupilot\Tools\AP_Periph\hwing_esc.cpp

**Version**
4.2.1

**Platform**
[X  ] All

**Airframe type**
All

**Hardware type**
Any ChibiOS platform.

**Logs**
NA

peterbarker, 2022-06-12:
Thanks for this, @pbreed .  By a curious coincidence this very issue came up on a recent DevCall.  We kind of think of `read()` as always returning in the range 0-255, even if the API - and even the codepaths! - allow for it to be outside those ranges.  That discussion was prompted by this: https://github.com/ArduPilot/ardupilot/pull/20901/files#r890661976

Note that in the vast majority of cases `read()` *can't* return -1 as `available()` has checked how many bytes are available in the relevant RingBuffer object and nothing else is pulling data from that RingBuffer.

We discussed the issue and thought it might even be worth changing read() to return `uint8_t`, squashing all errors to 0.  This would save a chunk of flash space (and boilterplate code) where we cope with the possible -1 return value.

Personally, I *loath* the `read()` call.  I far prefer passing in a buffer to be filled.  I've got branches floating around here trying to eliminate `read()`....

pbreed, 2022-06-12:
Actually that is not true. Available is not thread owner checked, read is.
So the bug I was chasing there WAS data available but read failed ANYWAY. 
Sticking the task in an endless loop trying to read data it could never actually read.

andyp1per, 2022-06-12:
> Actually that is not true. Available is not thread owner checked, read is. So the bug I was chasing there WAS data available but read failed ANYWAY. Sticking the task in an endless loop trying to read data it could never actually read.
> 
> Also a lot of over the wire protocols need to be able to read a 0 so folding -1 to zeros is still broken.

I think probably the solution is to fix available() otherwise we will keep getting in this situation

pbreed, 2022-06-12:
It is possible this will have ripple effects as I believe availible is labeled as const 
and not sure the functions from the OS to get the current thread wont make the compiler complain.
(Frankly that function should be const too, but waaaay under the covers the OS has some C functions 
so it may not be clean all the way to the bottom of the world.)

andyp1per, 2022-06-12:
> It is possible this will have ripple effects as I believe availible is labeled as const
> and not sure the functions from the OS to get the current thread wont make the compiler complain.
> (Frankly [...]

Works fine - https://github.com/ArduPilot/ardupilot/pull/20963

Can you try it with your problem?

closed 2022-07-13 by tridge