ArduPilot/ardupilot #20962: Hover on failsafe option

Angus-TStop, 2022-06-12:
## Feature request

**Is your feature request related to a problem? Please describe.**
When flyinig in wooded areas the current failsafes are not safe. Clearly RTL is no good but land can often be as bad due to the forest/wood floor only being suitable for landing in sporadic locations.

**Describe the solution you'd like**
Failsafe option of 'hover'

**Describe alternatives you've considered**
Condisdered the other failsafe options but not sure how they would work safely in such environments.

**Platform**
[  ] All
[  ] AntennaTracker
[Y] Copter
[Y] Plane (eVtols)
[  ] Rover
[  ] Submarine

**Additional context**
I fly a lot on film and TV sets and often the RTL option is a non starter due to overhead obstructions and landing is not an option either due to safety of film crew/sets/film equipment/ground directly under the drone

IamPete1, 2022-06-12:
For copter a option to switch in to brake mode would accomplish this.

What sort of failsafe? Holding position is really only a short term solution.... I guess it might give you chance to re-position your ground side transmitter/GCS link.

rmackay9, 2022-06-13:
Txs for the report/enhancement request.

This is probably related to and/or a duplicate of the request to add a two-stage failsafe https://github.com/ArduPilot/ardupilot/issues/16783

Angus-TStop, 2022-06-13:
Yeah, its definitely a temporary thing. The idea is to either move closer to the aircraft to regain the link or allow enough time for people to move out of the way (if on a film set for example).

Angus-TStop, 2022-06-13:
Its not really a duplicate of the other threads I don't think. This is just a hover option that remains until signal is regained regardless of time. That being said if, as per the other thread, break mode can be activated for a set time but the time can be set to 1-2 mins then that would do. One of the reasons I ask is that a hover failsafe saved the day on one occassion in my days of flying, a DJI A2 aircraft (before I saw the light of Arducopter!) and the long hover allowed us to clear the crew out and to regain lost link- moving closer and getting a crew member to turn their kit off.