ArduPilot/ardupilot #20985: Gimbal enhancement list

rmackay9, 2022-06-16:
This is a consolidated list of gimbal enhancements requested during development of the [Gremsy gimbal driver](https://github.com/ArduPilot/ardupilot/pull/20915).

- [ ] health reporting and pre-arm check
- [ ] extend 60s timeout for discovering a gimbal.  Perhaps allow the gimbal to be discovered as long as the vehicle is disarmed
- [ ] DO_GIMBAL_MANAGER_PITCHYAW mission and immediate command should support
    - [ ] both earth-frame vs body-frame angles (currently always earth-frame)
    - [ ] earth-frame and body-frame rates (how can we do this?  users would need to add NaN to angles?)
- [ ] add GIMBAL_MANAGER_SET_PITCHYAW message support (not to be confused with MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW)
- [ ] respond to requests for [GIMBAL_MANAGER_INFORMATION](https://mavlink.io/en/messages/common.html#GIMBAL_MANAGER_INFORMATION)
- [ ] support sending GIMBAL_MANAGER_STATUS (perhaps only if message-interval has been used to request it be sent)
- [ ] consume [GIMBAL_MANAGER_SET_ATTITUDE](https://mavlink.io/en/messages/common.html#GIMBAL_MANAGER_SET_ATTITUDE) messages
- [ ] bring other gimbal drivers up to the same level as the Gremsy driver including
    - [ ] lock vs follow
    - [ ] consume attitude information from gimbal
    - [ ] health reporting

These ground stations Issues and PRs should also be resolved/merged

- [ ] [MissionPlanner PR : mavcmd: add DO_GIMBAL_MANAGER_PITCHYAW command](https://github.com/ArduPilot/MissionPlanner/pull/2882) 
- [ ] [MissionPlanner Issue : Plan: ATTITUDE_TIME climb rate field reset to zero](https://github.com/ArduPilot/MissionPlanner/issues/2885)
- [ ] [MAVProxy Issue : WP Editor corrupts command's Param5 if not Latitude ](https://github.com/ArduPilot/MAVProxy/issues/1069)

olliw42, 2022-06-16:
fantastic list

- on the very top of the list I would put removing the need for the private mavlink channel thing (I'm not sure that's still required, but the discourse thread makes me believe so)
- I also would put on high priority sending the proper estimator flags in AUTO_PILOT_STATE_FOR_GIMBAL_DEVICE. Currently you simply send your ArduPilot specific EKF flags, which is really kind of rude, and outside of any standard.
- generally: get the communication in line with gimbal v2
- especially: removal of any old MOUNT_xxx stuff in gimbal v2 based gimbal drivers (like gremsy), including the modes stuff
- concerning gimbal discovery time: I indeed do it that way in BetaCopter/Plane, to link it to armed/not armed
- support also in other vehicles, especially plane
- sending AUTO_PILOT_STATE_FOR_GIMBAL_DEVICE only to the gimbal, and not system-wide broadcast
- support of landing state flag in AUTO_PILOT_STATE_FOR_GIMBAL_DEVICE, for all vehicles, also plane

ground station/MP issues
- support of GIMBAL_DEVICE_ATTITUDE_STATUS, i.e. fill campointa/b/c from this
- support of GIMBAL_MANAGER_INFORMATION, if it finds a gimbal heartbeat it should request it, and display the info in some useful location
- make it behave nicely with mavlink components, e.g. when they do not have parameters, or do not support other things a fc appears to be implicitly expected to support, like AUTOPILOT_VERSION, etc (see also https://github.com/ArduPilot/MissionPlanner/issues/2672).

:)

rmackay9, 2022-06-23:
@olliw42,

Txs very much for the list.  By the way, I've created another PR which (I hope) takes us another step forward.  Also I think I need hardware to test all our various drivers so I've [raised a proposal including for a SToRM32 gimbal](https://discuss.ardupilot.org/t/gimbal-hardware-for-randym/87220).  Any feedback or suggestions are greatly appreciated.

olliw42, 2022-06-30:
@julianoes 

> If I could add one item to the list, it would be to change the name of the driver to "MAVLink Gimbal v2" rather than Gremsy as the goal would be to have this as a generic protocol rather than a Gremsy specific one

there had been some discussion on this here: https://discuss.ardupilot.org/t/gimbal-hardware-for-randym/87220/5

Let me cite the relevant part from Randy's post: 
"I certainly like the idea of a single MAVLink gimbal driver but so far I’ve seen 3 different mavlink implementations (STorM32, Gremsy, NextVision) and they’re all different enough that I fear I would be unable to actually make it work. So for now I think I’m going to stick with separate drivers and hopefully the day will come where we can make them all inherit from the same base class."

rmackay9, 2022-07-01:
@julianoes,

Yes, what @olliw42 says is the current state.  I'm not opposed to it, I just need at least one more example gimbal that can use the driver as-is.  Instead of renaming the current one, I would probably create a new one (maybe called "MAVLink Gimbal v2" that uses the same driver under the cover but who knows at this point.

Txs very much for the offer re the gimbalv2 docs.

We still have a couple more messages to add support for and then I was thinking of removing the WIP tags on at least the messages that we're using in AP.

rmackay9, 2022-07-20:
@julianoes,

I've realised that the new mavlink messages don't have the concept of [MAV_MOUNT_MODE](https://mavlink.io/en/messages/common.html#MAV_MOUNT_MODE).  At the moment, we're continuing to support the [DO_MOUNT_CONTROL](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_MOUNT_CONTROL) and DO_MOUNT_CONFIGURE commands that allow setting the mode but I wonder how you were thinking we could do without the mode concept and the ability to set it?

In particular I think the pilot (or mission or lua script) needs to be able to specify the gimbal go into special modes like the Retracted position (which I'm interpreting as the gimbal relaxing the motors) or Neutral position (which is normally pointing forward).  These two seem to be handled using the the [GIMBAL_MANAGER_FLAGS](https://mavlink.io/en/messages/common.html#GIMBAL_MANAGER_FLAGS) which has neutral and retract bits.

.. but the pilot also wants to be able to specify that they're in control using the RC transmitter which they do by setting the mode to RC_TARGETING.  I suspect you will say that we should use [MAV_CMD_DO_GIMBAL_MANAGER_CONFIGURE ](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_GIMBAL_MANAGER_CONFIGURE) but "RC input" is not a mavlink system so it doesn't have a sysid/comp-id.

.. there are other gimbal modes that we support including pointing at home, pointing at a sysid (e.g. point at another vehicle), etc.

julianoes, 2022-07-20:
> These two seem to be handled using the the [GIMBAL_MANAGER_FLAGS](https://mavlink.io/en/messages/common.html#GIMBAL_MANAGER_FLAGS) which has neutral and retract bits.

Yes, that was the idea there.

> but "RC input" is not a mavlink system so it doesn't have a sysid/comp-id.

So to signal that RC is in control, I use the sysid_primary_control and compid_primary_control of the autopilot (for PX4 that's by default 1, 1) which means that itself is in control, and itself has an RC receiver. I understand this is a bit of a kludge...

> pointing at home, pointing at a sysid (e.g. point at another vehicle)

For tracking we usually would set some camera with sysid/compid in control which then does the tracking but I can see how that doesn't answer pointing at a sysid and pointing at home.

Should we extend [MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW) for these use cases or should we add one more message, e.g. MAV_CMD_DO_GIMBAL_MANAGER_TRACK, and then an enum listing the various tracking options.

And that would be different from [MAV_CMD_CAMERA_TRACK_POINT](https://mavlink.io/en/messages/common.html#MAV_CMD_CAMERA_TRACK_POINT) which is more for a camera.

olliw42, 2022-07-20:
@julianoes 

this isn't meant sarcastic at all, but you may understand that I kind of feel confirmed in what I am saying since ca 2 years and not getting through with it
- maybe you want to finally revise your position concerning RC targeting/control, not only for gimbal device but also gimbal manager
- maybe you want to think through what I call "super modes" and what is essentially mount mode, only put onto a consistent and extensible concept

while your suggestions may somehow may allow to achieve the things, it's obviously of the kind of from-behind-through-the-chest type of workaround. As I always do, I wonder, why not putting it onto solid feet

sorry for being a pain

> So to signal that RC is in control, I use the sysid_primary_control and compid_primary_control of the autopilot (for PX4 that's by default 1, 1) which means that itself is in control, and itself has an RC receiver. I understand this is a bit of a kludge...

and there might be other reasons it may be set to itself in control, like missions, .... how then to discriminate what the intention is

> For tracking we usually would set some camera with sysid/compid in control which then does the tracking but I can see how that doesn't answer pointing at a sysid and pointing at home.

but most cameras out there can't be set with sysid/compid in control and then do the tracking, so not even this would be an answer in many cases

> Should we extend [MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW) for these use cases or should we add one more message, e.g. [...]

so you end up with a message/command for every case even though all it needs is a mount_mode (aka super mode) value

hmhmhm

rmackay9, 2022-07-20:
@julianoes @olliw42,

OK great thanks.  I'll go and think about it a bit more but I think we need to retain the concept of gimbal mode rather than trying to squish it into the existing messages or flags.   I think we will probably need a set-mode command and also reporting back somehow from the gimbal manager on what the gimbal's mode is.

olliw42, 2022-07-20:
> I think we need to retain the concept of gimbal mode rather than trying to squish it into the existing messages or flags. I think we will probably need a set-mode command and also reporting back somehow from the gimbal manager on what the gimbal's mode is.

that's my opinion too, except that I made "gimbal mode" a bit more general, calling it "super mode" (which is admitedly a silly name, pl feel free to invent a better one).

I wonder if you might find it usefull to read the part on "super modes" in here http://www.olliw.eu/2020/mavlink-gimbal-protocol-v2/. Sadly, there is no direct link to the relevant location, so please scroll down to the paragraph which starts with "Remains to clarify what the super modes are."
(seraching in the web page for super modes should lead you quickly)

I wonder if this isn't exactly what you actually want.

julianoes, 2022-07-20:
Alright, I'm open for suggestions :smile:.

KylePreuss, 2023-02-09:
Without the concept of these modes how does a GCS know if the gimbal is supposed to be looking at a lat/lon (i.e. MAV_MOUNT_MODE_GPS_POINT) vs being controlled by angles or slews? It sounds like it could be accomplished thru additional GIMBAL_MANAGER_FLAGs or a mode field. Either way it may be useful.

Along those lines I'm curious how the GCS might request / see the last commanded values for those modes e.g. get the lat/lon from the last MAV_CMD_DO_SET_ROI_LOCATION but that is another issue.

@olliw42 FYI your link on the gimbal protocol has gone down

thespova, 2026-01-13:
Is this the right place to suggest that all gimbals including servo gimbals should have an FPV mode option?  Seems simpler to support this with servo gimbals than any others since its less functionality than is already in ardupilot with the AP doing the pitch leveling now.

thespova, 2026-01-13:
> Is this the right place to suggest that all gimbals including servo gimbals should have an FPV mode option? Seems simpler to support this with servo gimbals than any others since its less functionality than is already in ardupilot with the AP doing the pitch leveling now.

I see that @Hwurzburg  has already setup the servo gimbal type to use the same params as the gm gimbals. Not sure where PRs are for it but on the bench this all seems to work great and exactly like I hoped for servo gimbals.