ArduPilot/ardupilot #20995: Copter: unable to arm when throttle is maximized in GUIDED mode

muramura, 2022-06-17:
## Bug report

**Issue details**

I did an ARM from a QGC in GUIDED mode with the throttle of the radio on top.
I could not arm with "PreArm: Throttle (RC3) is not neutral".
I expect it to ignore the radio input except for the flight mode channel.
I can transition from GUIDED mode to other modes by changing the flight mode channel.

**Version**

ArduCopter V4.3.0-dev (e9942f9a)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

QUAD COPTER

**Hardware type**

SITL]

**Logs**

[image]

muramura, 2022-06-17:
I would add a decision not to check the radio input if the arm is by MAVLINK.
With this addition I could arm ignoring the propo stick.

[image]

rmackay9, 2022-09-16:
We have a related (perhaps duplicate?) issue here https://github.com/ArduPilot/ardupilot/issues/21415

peterbarker, 2024-09-10:
We have a bit in RC_OPTIONS that handles this now.