ArduPilot/ardupilot #21007: Copter: Parachute can be manually released when climbing

GiantSeaweed, 2022-06-20:
## Bug report

**Issue details**

Parachute can be manually released when the Copter is climbing. 
I notice that Parachute_manual_release() only check the altitude and land mode.
I find a related post here about parachute (manual) release condition: https://discuss.ardupilot.org/t/parachute-release-conditions/56563

**Version**

Copter 4.2.1

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

QuadCopter

**Hardware type**

SITL

**Logs**
The commands to reproduce this behavior:
> parachute enable
> mode GUIDED
> arm throttle
> takeoff 20
> parachute release

[image]

IamPete1, 2022-06-20:
The parachute is for when things are going wrong, we don't want assume we know better than the pilot in that case. For example what if the copter thinks its climbing but is not.

IamPete1, 2022-06-20:
Right, that the parachute is having to be deployed something must have gone wrong. It could be that AP's state estimation has gone wrong. The pilot is in the position to make that judgment call where the flight code is not.

Of course you could make the same argument about the altitude check, but that is already there....

rmackay9, 2022-06-20:
looks like we have an answer so I might close this.  As IamPete1 says, we generally try to keep AP doing what it's told and not try to second guess what the pilot wants.  If the pilot says he/she wants to release the parachute, we should generally do that unless we're pretty sure it's a mistake.

closed 2022-06-20 by rmackay9