ArduPilot/ardupilot #21043: SITL Compile Failed on building AP_Soaring.cpp

KrishnanS2006, 2022-06-26:
## Bug report

**Issue details**

Compiling SITL fails, with the below error:
```cpp
../../libraries/AP_Soaring/AP_Soaring.cpp: In constructor ‘SoaringController::SoaringController(AP_TECS&, const AP_Vehicle::FixedWing&)’:
../../libraries/AP_Soaring/AP_Soaring.cpp:171:17: error: member ‘SoaringController::_polarParams’ is used uninitialized [-Werror=uninitialized]
  171 |     _speedToFly(_polarParams),
      |                 ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

Waf: Leaving directory `/home/krishnan/Coding/TJHSST/TJUAV/ardupilot/build/sitl'
Build failed
 -> task in 'objs/AP_Soaring' failed (exit status 1): 
        {task 140141097210976: cxx AP_Soaring.cpp -> AP_Soaring.cpp.0.o}
 (run with -v to display more information)
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks
```

Some additional information: I am on Arch, and all packages (including `gcc`) are completely up-to-date (after a `sudo pacman -Syu`). The exact configuration was working perfectly for the past few months, until SITL decided to re-compile the files again when running `ardupilot/Tools/autotest/sim_vehicle.py --no-mavproxy -L CUSTOM_LOCATION -v ArduPlane --add-param-file ardupilot/Tools/autotest/default_params/custom.parm`

**Version**
I am currently on `origin/Plane-4.2` on commit `44735215e5bf4dc26b55c1c3b0d361188c2927d1` in order to keep firmware consistent. However, I have tried with the latest `origin/master` and got the same error

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Plane (Configured with `./waf plane`)

**Hardware type**
The board was configured as `./waf configure --board sitl --debug` to use SITL.

acxz, 2022-07-12:
Maybe this is helpful: https://stackoverflow.com/questions/61240589/how-to-initialize-a-struct-to-0-in-c

But I'm not sure how to create an `AP_Float`.

acxz, 2022-07-13:
@khancyr do you have any suggestions for resolving this issue?

I can't seem to create an `AP_Float`, and besides that the only other solution I see is to remove the usage of the struct and replace it with a `AP_Vector3f`

khancyr, 2022-07-14:
@acxz fixed in https://github.com/ArduPilot/ardupilot/pull/21183

acxz, 2022-07-14:
Thanks so much!

closed 2022-07-20 by tridge