ArduPilot/ardupilot #21053: 4.2 Stable ASL instead of AGL data for the tracker missing

jaroszmm, 2022-06-27:
## Bug report

When feeding my tracker data using Mavlink Emu with crossfire 6.10 firmware on 4.2 stable I am getting ASL altitude data instead of AGL which makes the tracker unusable

**Version**
Plane 4.2 stable

**Platform**
Plane

andyp1per, 2022-06-27:
Mavlink Emu changed a lot in 6.13, I think you should first upgrade and see whether it's any better

jaroszmm, 2022-06-28:
Don't think it's Crossfire.
Friend made me a custom build sometime ago based on 4.1 dev and he manually swapped ASL with AGL so my other planes on Crossfire 6.10 work fine.

IamPete1, 2022-06-28:
@jaroszmm Can you get a TLog of the MAVLink data from the Crossfire. https://ardupilot.org/plane/docs/common-mission-planner-telemetry-logs.html

jaroszmm, 2022-06-30:
> @jaroszmm Can you get a TLog of the MAVLink data from the Crossfire. https://ardupilot.org/plane/docs/common-mission-planner-telemetry-logs.html

I can, next time I go flying I'll get my tracker with me, make a short flight with tracking and bring the log here.