ArduPilot/ardupilot #21058: Please tell me what's wrong with my log

GWANHO1, 2022-06-28:
Recently, the aircraft crashed during flight. I've analyzed the logs for this aircraft, but I'm not sure what the problem is. please let me know  

**Issue details**

_Please describe the problem_
During the flight, the controller could not be controlled and it was moving and crashed.

**Platform**
[  ] All
[  ] AntennaTracker
[ ㅇ ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
 Quad Copter X TYPE
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk 4
**Logs**
_Please provide a link to any relevant logs that show the issue_
[2022-06-02 17-23-50.bin0wp.txt](https://github.com/ArduPilot/ardupilot/files/8997262/2022-06-02.17-23-50.bin0wp.txt)
[2022-06-02 17-31-30.bin0wp.txt](https://github
[2022-06-02 17-23-50.bin0wp.txt](https://github.com/ArduPilot/ardupilot/files/8997269/2022-06-02.17-23-50.bin0wp.txt)
.com/ArduPilot/ardupilot/files/8997265/2022-06-02.17-31-30.bin0wp.txt)

GWANHO1, 2022-06-28:
[2022-06-02 17-23-50.zip](https://github.com/ArduPilot/ardupilot/files/8997271/2022-06-02.17-23-50.zip)
[2022-06-02 17-31-30.zip](https://github.com/ArduPilot/ardupilot/files/8997273/2022-06-02.17-31-30.zip)
[2022-06-02 19-36-30.zip](https://github.com/ArduPilot/ardupilot/files/8997274/2022-06-02.19-36-30.zip)

amilcarlucas, 2022-06-28:
Pleased post support questions on the ArduPilot forum. Here only confirmed bugs are accepted.

closed 2022-06-28 by amilcarlucas