ArduPilot/ardupilot #21080: Copter: DO_CHANGE_SPEED not working if immediately after a TAKEOFF command

rmackay9, 2022-06-30:
As reported here a DO_CHANGE_SPEED command does not take effect if it is placed immediately after a TAKEOFF command.

[image]

Placing the do-change-speed after a waypoint command does work though
[image]

The issue from the first item must be that WPNav's init() is being called after the do-change-speed command is executed.  This "init" then resets the speed to the default.

This bug was probably introduced when we changed takeoff to use the position controller.  FYI @lthall

rmackay9, 2022-09-19:
Another variation of this issue appeared [here in the forums](https://discuss.ardupilot.org/t/mission-ignoring-do-change-speed-flying-wpnav-speed-instead/90815/2) and involves the do-change-speed not taking effect even if after a NAV_DELAY command.

prathamesh7255, 2023-02-20:
@rmackay9 sir, should we change this feature so that even if the DO_CHANGE_SPEED is set before deciding the first waypoint, it will store the speed, and after determining the first waypoint, it will fly with the set speed? Or should we allow users to set the speed only after deciding on the first waypoint?

rmackay9, 2023-02-20:
@prathamesh7255, we should support the do-change-speed even if before the first waypoint.. I think this will mean storing the speed in the mode_auto class and then setting the wpnav object's speed when we start using it.

vinny-ferrari, 2023-04-17:
@prathamesh7255 I just compiled a version 4.3.6 for CubeOrange+ with your modification and it works fine! I will do more tests but, everthing looks ok! Thank You!

rmackay9, 2023-09-15:
We had a report from a Partner company today (flying 4.2) which showed a slight variation of this bug.  It seems that UGCS places a do-change-speed command before the takeoff.  This results in the vehicle ignoring the do-change-speed if Auto mode is started while the vehicle is on the ground (or more accurately while the vehicle is below the takeoff command's target altitude) but the command does take effect if the vehicle is switch to Auto when the vehicle is already above the takeoff altitude.

rmackay9, 2024-03-15:
This has been fixed in 4.5.0 which is now in beta so closing.

closed 2024-03-15 by rmackay9

heesup, 2025-09-05:
Hello, this issue still occurs in Arducopter V4.5.6.

Even when I use the same waypoint file, the problem happens when the copter starts the mission from the air instead of being launched from the TAKE_OFF point. My assumption is that when the copter starts the AUTO mode and is already in the air, it will GOTO to WAYPOINT (skipping the DO_CHANGE_SPEED between TAKEOFF and WAYPOINT), which causes it to ignore the speed change command. I'm not familiar with the Ardupilot code, but I believe it's related to this code file.

https://github.com/ArduPilot/ardupilot/blob/07cd29fce8f22bc1a667e5434f3860e40c177452/ArduCopter/mode_auto.cpp#L1603

Please ignore this comment if the issue is already resolved. I attached the mission file just in case it's helpful. (Also, a flight log later if it's needed)

[2025-08-12-2m-part3.waypoints.zip](https://github.com/user-attachments/files/22177468/2025-08-12-2m-part3.waypoints.zip)