ArduPilot/ardupilot #21086: WeBots Hexacopter Example

wrmichaelmak, 2022-06-30:
## Feature request

**Is your feature request related to a problem? Please describe.**
I am trying to get a hexacopter running with SITL in WeBots. I tried to adapt the quad controller code to do accommodate 6 motors but it does not seem to be working.

**Describe the solution you'd like**
I want to be able to run SITL with a hexacopter on WeBots.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

HefnySco, 2022-07-05:
what version of webots are you using ?

HefnySco, 2022-07-06:
Please check this link [https://github.com/HefnySco/ardupilot/tree/pr_SITL_webots_2022a](https://github.com/HefnySco/ardupilot/tree/pr_SITL_webots_2022a) it contains the updated version.
If you would like you could add your hex as model as well.

HefnySco, 2022-07-06:
WeBots 2022A coordinates is changed from Webots2021 that is why there is no floor and it will not fly the drone on 2022A.
the link above contains the updates -in progress version- Quad is working well.

closed 2022-07-06 by HefnySco

wrmichaelmak, 2022-07-06:
Sure I will add it in the future. Will it be worth making the models compatible with ENU instead of NUE considering the fact that a lot of robotic framework/simulator is using ENU.