ArduPilot/ardupilot #21098: Copter: circle mode can't point vehicle at center if CIRCLE_RATE > ATC_SLEW_YAW

rmackay9, 2022-07-02:
As discovered in this [discussion](https://discuss.ardupilot.org/t/strange-behaviour-in-circle-mode/87110/10) if the CIRCLE_RATE is higher than ATC_SLEW_YAW then the vehicle's yaw will not be able to keep pointed at the circle's center.  Most of the time the CIRCLE_RATE is much lower than the ATC_SLEW_YAW so this hasn't been noticed before.

We should probably do one of the following:

1. add a note to the CIRCLE_RATE parameter description <-- I think I like this idea best
2. limit the CIRCLE_RATE to be no more than ATC_SLEW_YAW (I'm not a fan of this idea)
3. print a warning to the user when they enter circle mode for the first time that CIRCLE_RATE is too high
4. add a pre-arm check in the parameter section

timtuxworth, 2022-07-02:
Number 3 and/or 4 could be implemented in Lua script - so they can be enabled/disabled without being baked into the firmware. I have a draft PR for this concept here https://github.com/ArduPilot/ardupilot/pull/20930

mohitkenia7, 2022-11-12:
Hi 
I want to work on this

mohit-marathe, 2022-12-12:
Hey @rmackay9, I'm new here and want to work on this issue. Can you help me with where I should get started?

Bhawna1203, 2022-12-23:
Hii,
I am working on this issue

PR #22720 (2023-01-23): Copter: AC_WPNav: circle_rate param description added
closes #21098 
added parameter description in Circle_rate that it must be smaller than Slew_yaw. Open for feedback.
thanks to @rmackay9 and @IamPete1 for the help. I found what I was doing wrong and will avoid it in future.
closed 2023-01-23 by rmackay9