ArduPilot/ardupilot #21101: FMUV6X/FMUV5X: Ethernet is not available

muramura, 2022-07-02:
## Feature request

**Is your feature request related to a problem? Please describe.**

FMUV5X and FMUV6X implement Ethetnet devices.
RASPBERRY PI and FC are connected via UART.
RASPBERRY PI uses MAVPROXY and the internal application uses TCP/IP for telemetry communication.

**Describe the solution you'd like**

If FC Ethernet communication is available, faster communication than UART is possible.
Also, if ETHERNET communication is available, we can build an intranet for drones.
Please support Ethernet communications.

**Describe alternatives you've considered**

None

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
[image]
[image]

timtuxworth, 2022-07-02:
This is my blog post about UARTs - which asks the question, why are we not using more modern, reliable, faster, bus based communications like Ethernet (among others). https://timtheplaneman.com/uarts-on-flight-controllers-huh/

muramura, 2022-07-02:
UART is specified for connection to companion computers and telemetry devices.
These devices are connected at 921600 bps or faster.
115200 bps was a few years ago.

Companion Computers
https://ardupilot.org/dev/docs/companion-computers.html

rmackay9, 2022-07-04:
Yes, we have work that's ongoing to add ethernet support to ArduPilot for the STM32 based autopilots (the Linux flight controllers already support ethernet).  I suspect that will be included in 4.3 although I can't 100% guarantee that.

muramura, 2022-07-04:
Thanks for the comment, I appreciate it.

magicrub, 2022-07-09:
The Ethernet code that @bugobliterator and I are working on have significant changes. I don't think it will make it into 4.3 in time.

timtuxworth, 2022-07-09:
It will be great to have it whenever it's ready!

Ryanf55, 2023-03-23:
> The Ethernet code that @bugobliterator and I are working on have significant changes. I don't think it will make it into 4.3 in time.

Do you think it could make 4.5 so it can be used in DDS by #23283?

peterbarker, 2024-05-07:
We have ethernet support now.

closed 2024-05-07 by peterbarker