ArduPilot/ardupilot #21114: Serial passthrough is broken

andyp1per, 2022-07-04:
## Bug report

Following the serial passthrough guidelines on the wiki for GPS does not work on 4.2.x or 4.3.x. A connection can be established but no data appears to flow. 

**Version**
4.2.x, 4.3.x

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

mbuzdalov, 2022-08-12:
Seems to be there for quite a while. With a JHEMCU GSF405A board, I was trying to flash a ELRS module, which is hardwired to SERIAL1, using passthrough, to no success. I though this is me who fails, but apparently not.

andyp1per, 2022-08-12:
@bugobliterator any update on this?

teodoryantcheff, 2022-11-23:
Anything?

teodoryantcheff, 2023-05-20:
Almost a year later, have any changes been done to the serial passthrough code?

rmackay9, 2023-05-21:
I've added this to the Copter-4.4 issues list so it won't be forgotten.

I think serial pass through works in some cases at least because I used it a few weeks ago... so I suspect this issue is just for GPSs.

teodoryantcheff, 2023-05-22:
Thank you, @rmackay9 !
I think this is not limited to GPSes only, since as you can probably see in the "attached" video, back when I was testing this, a "GPS" is completely not involved in reproducing the issue reported. 
Or, of course, there's something that I may be completely missing in the bigger picture.
Anyway -- many thanks, again!

rmackay9, 2023-05-25:
So this really is a problem and [we've got a fix](https://github.com/ArduPilot/ardupilot/pull/23888) for master (aka "latest") that we will backport to 4.4 for sure and possibly 4.3 if we ever do another release of 4.3.

rmackay9, 2023-06-05:
@andyp1per,

How do you feel about closing this now?

andyp1per, 2023-06-05:
I haven't tried it but I trust @tridge - happy to close

closed 2023-06-05 by andyp1per