ArduPilot/ardupilot #21125: Altitude Fence: Use the ground height of the current position for altitude fence determination

muramura, 2022-07-06:
## Feature request

**Is your feature request related to a problem? Please describe.**

The altitude at which a vehicle can fly freely in Japan is less than 150 meters above ground level at the current location.
The maximum altitude of the altitude fence is the ground level of the takeoff position.
Japan is a country with few plains. The ground surface in Japan Country is up and down.
For example, a vehicle is flying horizontally at an altitude of 149 meters above the takeoff position.
During the flight, the ground surface is 2 meters lower.
At this time, the vehicle is flying at an altitude of 151 meters above the ground.
In this case, the vehicle is in violation of the law.
I wish to obey the law.

**Describe the solution you'd like**

I want to check the altitude fence at the ground level of my current position.

**Describe alternatives you've considered**

I want Location::AltFrame::ABOVE_HOME to be Location::AltFrame::ABOVE_TERRAIN.
Or make it selectable by config parameter.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
[image]

timtuxworth, 2022-07-06:
This would be good for Plane too. Canada is also has similar issues. In many places.

rmackay9, 2022-07-07:
Yes, this is definitely a good idea.  It is not quite so simple to implement as you might first expect though.  It would mean that the altitude fence requires position which it does not currently and we also need to handle pushing the vehicle down as it moves horizontally.  At the moment in Loiter mode (and other manual modes) the vehicle simply stops climbing if the pilot tries to push it above the altitude fence.  We would need to enhance this to actually push the vehicle down in some cases.

muramura, 2022-07-07:
@rmackay9 san. Comments thanks.