ArduPilot/ardupilot #21137: Increase Boardname Length & Output Fields for Vehicle Manufacturer's in Banners

hendjoshsr71, 2022-07-07:
## Feature request

We have received a number of requests from Partners and an issue on Discuss ([here](https://discuss.ardupilot.org/t/zealoth743-board-name-truncated-in-gcs-messages/86770/9)) regarding the 13-char limit on board names. 

Code here: [Link](https://github.com/ArduPilot/ardupilot/blob/1087b297ac14885857bdc81f87c95f7535940b09/libraries/AP_HAL_ChibiOS/Util.cpp#L362)

@peterbarker suggested one possibility of no longer sending the MCU ID to have more chars available. As it is available in `AUTOPILOT_VERSION`. 

But has the concern that doing this may break automated tooling or GCSs looking for that format. 

I do think MCU_ID has value in the banner though.... it does get used to confirm we are looking at the same vehicle as before.

**Vehicle Manufacturer Name & Vehicle Model Name**
In a similar area I would like to add something to output `Vehicle_Manufacturer_Name` and `Vehicle_Model` as part of the GCS banner.  Via a define when creating firmware with parameters, etc. baked in.

In the past I've added just shortened the autopilot name, but that isn't great for support engineers who receive logs a lot ;) Nor is it as clear as full names.

WickedShell, 2022-07-07:
The MCU ID should be kept unless there is a separate way that ensures it shows up in the dataflash log. I also do think it's useful for vehicle differentiation as you said. The main thing for me is just ensuring that we can see the ID in both the DF and tlogs.

peterbarker, 2022-07-12:
- emit MCU ID in `MSG`
 - also emit in VER dataflash message
 - stop emitting as statustext
 - add sending autopilot_version to send_banner