ArduPilot/ardupilot #21138: Allow different power allocation between top and bottom motors for co-axial drones

eduardkieser, 2022-07-07:
**The problem/challange**
When setting up co-axial drones like OctoQuad it is usefull to make the bottom motors turn faster since they are generally not working as hard as the top motors. In [this thread](https://discuss.ardupilot.org/t/new-build-x8-top-bottom-props-recommendation/20333/27) users describe buying different KV motors, using different props and modifying ESC calibration to do this.

**One naive solution**
It would be super cool if we could get something like a `COAX_BOTTOM_BOOSTER` 
Where, 
`bottom_motor_output = clip(bottom_motor_output*COAX_BOTTOM_BOOSTER)`

In the hope that setting this value to 1.2, would make the system send approximately 20% more power to the bottom motors without causing the thing to fall out of the sky.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Additional context**
I posted 2 other somewhat related questions in the wiki, [here](https://discuss.ardupilot.org/t/co-axial-motor-power-allocation-for-efficiency-improvement/87918) and [here](https://discuss.ardupilot.org/t/co-axial-motor-power-allocation-to-improve-yaw-response/87915)

IamPete1, 2022-07-07:
This is possible using scripting matrix and the `set_throttle_factor` binding. Script would be something like:
```
-- duplicate the #defines from AP_Motors
local AP_MOTORS_MATRIX_YAW_FACTOR_CW = -1
local AP_MOTORS_MATRIX_YAW_FACTOR_CCW = 1

local AP_MOTORS_MOT_1 = 0
local AP_MOTORS_MOT_2 = 1
local AP_MOTORS_MOT_3 = 2
local AP_MOTORS_MOT_4 = 3
local AP_MOTORS_MOT_5 = 4
local AP_MOTORS_MOT_6 = 5
local AP_MOTORS_MOT_7 = 6
local AP_MOTORS_MOT_8 = 7

-- helper function duplication of the one found in AP_MotorsMatrix
local function add_motor(motor_num, angle_degrees, yaw_factor, testing_order)

    MotorsMatrix:add_motor_raw(motor_num,math.cos(math.rad(angle_degrees + 90)),
                                         math.cos(math.rad(angle_degrees)),
                                         yaw_factor,
                                         testing_order)

end

-- this duplicates the add motor format used in AP_Motors for ease of modification of existing mixes
-- standard octa quad X mix
add_motor(AP_MOTORS_MOT_1,  45, AP_MOTORS_MATRIX_YAW_FACTOR_CCW,  1)
add_motor(AP_MOTORS_MOT_2, -45, AP_MOTORS_MATRIX_YAW_FACTOR_CW, 7)
add_motor(AP_MOTORS_MOT_3, -135, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 5)
add_motor(AP_MOTORS_MOT_4, 135, AP_MOTORS_MATRIX_YAW_FACTOR_CW, 3)
add_motor(AP_MOTORS_MOT_5,  -45, AP_MOTORS_MATRIX_YAW_FACTOR_CCW,  8)
add_motor(AP_MOTORS_MOT_6,  45, AP_MOTORS_MATRIX_YAW_FACTOR_CW,  2)
add_motor(AP_MOTORS_MOT_7, 135, AP_MOTORS_MATRIX_YAW_FACTOR_CCW,  4)
add_motor(AP_MOTORS_MOT_8, -135, AP_MOTORS_MATRIX_YAW_FACTOR_CW,  6)

-- setup throttle factors
-- top motors reduce throttle
set_throttle_factor(AP_MOTORS_MOT_1, 1/1.2)
set_throttle_factor(AP_MOTORS_MOT_2, 1/1.2)
set_throttle_factor(AP_MOTORS_MOT_3, 1/1.2)
set_throttle_factor(AP_MOTORS_MOT_4, 1/1.2)

-- bottom motors get full (the default is one so this lines don't actually do anything.)
set_throttle_factor(AP_MOTORS_MOT_5, 1.0)
set_throttle_factor(AP_MOTORS_MOT_6, 1.0)
set_throttle_factor(AP_MOTORS_MOT_7, 1.0)
set_throttle_factor(AP_MOTORS_MOT_8, 1.0)

assert(MotorsMatrix:init(8), "Failed to init MotorsMatrix")

motors:set_frame_string("Octo Quad with different top and bottom throttle")
```

rmackay9, 2022-07-08:
I think I'll re-open this enhancement request because it keeps coming up.  I think we can't really consider it resolved unless we write up on the wiki why it's not necessary or provide an easy way (with no lua) to resolve it.

Here are examples of the same request:

- https://github.com/ArduPilot/ardupilot/issues/665
- https://github.com/ArduPilot/ardupilot/issues/1467

IamPete1, 2022-07-08:
@rmackay9 Your happy with the scripting solution but what a Wiki page? We should close this and open a issue on the Wiki in that case. Or your not happy with Lua and want another solution, presumably parameters? We could make the Lua script generic and have it load the configuration params you need?

eduardkieser, 2022-11-30:
Hi @IamPete1 
Thanks for the help so far. I've been running into some issues though. The script (and all subsequent variants) fails when I try to do: `assert(MotorsMatrix:init(8), "Failed to init MotorsMatrix")` 
I've also tried the quad and hex examples on github and it always fails at the same point. I'm not sure if it can be related the the number of motors expected for my airframe type (octor quad X). `Q_FRAME_CLASS = 3` `Q_FRAME_TYPE = 1` to account for this I made another minimal example shown below, but with no luck. It still fails when I try to initialize the MotorMatrix. Any ideas? (Ive added a copy of my parameters in the txt file).

```
MotorsMatrix:add_motor_raw(0,-1, 0, 1, 2)
MotorsMatrix:add_motor_raw(1, 1, 0, 1, 4)
MotorsMatrix:add_motor_raw(2, 0, 1,-1, 1)
MotorsMatrix:add_motor_raw(3, 0,-1,-1, 3)

MotorsMatrix:add_motor_raw(4,-1, 0, 1, 5)
MotorsMatrix:add_motor_raw(5, 1, 0, 1, 6)
MotorsMatrix:add_motor_raw(6, 0, 1,-1, 7)
MotorsMatrix:add_motor_raw(7, 0,-1,-1, 8)

assert(MotorsMatrix:init(8), "Failed oct init..")

motors:set_frame_string("scripting plus example")
```

This keeps failing and causes the scripts engine to keep rebooting showing:

```
2022/11/30 12:04:24 PM : Failed o
2022/11/30 12:04:24 PM : PreArm: Scripting: /APM/scripts/quad_test.lua:11: 
2022/11/30 12:04:24 PM : PreArm: Radio failsafe on
2022/11/30 12:03:53 PM : PreArm: in RTL mode
2022/11/30 12:03:53 PM : Failed o
2022/11/30 12:03:53 PM : PreArm: Scripting: /APM/scripts/quad_test.lua:11: 
2022/11/30 12:03:53 PM : PreArm: Radio failsafe on
2022/11/30 12:03:22 PM : PreArm: in RTL mode
2022/11/30 12:03:22 PM : Failed o
```
[BC2_30_Nov.txt](https://github.com/ArduPilot/ardupilot/files/10121966/BC2_30_Nov.txt)