ArduPilot/ardupilot #21169: Internal errors 0x100000 l:917 flow_of_ctr on BeagleBone Blue

TilCreator, 2022-07-11:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_

Discussion: https://discuss.ardupilot.org/t/internal-errors-0x100000-l-917-flow-of-ctr/87033

## Bug report
I’m flying a quad copter with a BeagleBone Blue as a flight controller and Arducopter 4.2.1. After a few minutes of flying Arducopter reports error 0x100000 (flow_of_control) and doesn’t want to arm again after disarming: PreArm: Internal errors 0x100000 l:917 flow_of_ctr. It’s seams like the error is caused by “z_control” taking too long or maybe a time glitch.

**Issue details**
[Heli frame, prearm Internal error 0X100000, L:602 flow_of_ctr](https://discuss.ardupilot.org/t/heli-frame-prearm-internal-error-0x100000-l-602-flow-of-ctr/85172) and [Internal Error 0x100000 after update from 4.1.1 to 4.1.2](https://discuss.ardupilot.org/t/internal-error-0x100000-after-update-from-4-1-1-to-4-1-2/79684) seam possibly related, but not the same and should be fixed already.

[image]
I can also find the `PM.MaxT` value being quite high at the same time as the error message.

I would expect this error to not be triggered.

**Version**
4.2.1

**Platform**
- [ ] All
- [ ] AntennaTracker
- [x] Copter
- [ ] Plane
- [ ] Rover
- [ ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
BeagleBone Blue

**Logs**
https://gist.github.com/TilCreator/548cf61d1b633ea94ced6838c4f56acf
Error flag is triggered in line 210, it keeps reporting the error in the last lines.

IamPete1, 2022-09-04:
This is certainty the long loop times, I'm not really sure what we can do about this on the AP side. You could turn off some features I guess.

TilCreator, 2022-09-05:
How bad would it be to manually increase the timeout or not have to restart AP before arming again? (The restarting part is the most annoying part, because stuff like battery levels get reset)

IamPete1, 2022-09-05:
You had a loop equivalent of ~ 70hz on copter the target is 400hz by default. Its a genuine issue, the internal error is there for a reason.

I'm not really familiar enough with the Linux based flight controllers to suggest what you could do to speed it  up.