ArduPilot/ardupilot #21173: Separate Discord Server or Channels for Community Support?

hendjoshsr71, 2022-07-12:
Probably not the best place for this discussion, but an easy way to mark it for the call :)

Our Development Discord Server is great for helping new developers (me ;)) learn and get acquainted with code. And of course supports all of our many Partners!

We end up with many users in the server who rightfully think it should be for support too. See below.
[image]

The Discord link above should link to something for Users like the Facebook and Discuss links.

* Pro Separate Servers
- Easier management and not tangling it with our Development Server & Partner Channels
- Could get community members for admins easier without worrying about permissions as much.  
- Also support folks already have 120 channels and growing in one server :)

* Con
- Separates the community a bit. But we already do that anyways with directing users to Discuss
- Folks could still join both !

*Con to this all
- Discuss is a better platform for search engines and archiving previous answers.  IE we should encourage users to post to Discuss as well answers they found quite helpful for the next user or dev. 

*Pros to All
From @MichelleRos "While it would be nice to have it all Google-searchable, it's nicer for people to have somewhere to ask questions where we're not the only ones answering them. 
The format of Discord is just much easier to use than the forums - short messages that are like speaking, rather than big posts that feel more like writing a letter."

You may notice my preference is to make them different servers....

IamPete1, 2022-07-12:
I think we either need make it much clearer much that the discord is only for developers or accept that users want to use discord too and give them a place for that, either on our current server or on a new one. We do already have lots of paths for user support but at least its easy to mute channels on discord if its getting overwhelming. Hopefully we would also benefit from some of our more advanced users helping out with support as many do on the forum.

Trying to keep the current discord developer only as we try and do now is causing a increasing amount hassle on both sides.

peterbarker, 2022-07-16:
Thanks @hendjoshsr71 .  The stated goal is to try to keep the developer Discord clear of random support requests.  I don't think more channels are going to help there - I think we should go for a completely separate Discord server and advertise that on discuss.ardupilot.org

tridge, 2022-07-16:
i'd much rather use the same server and different channels, as it better aligns with the way discord works with "boosts"

MichelleRos, 2022-07-16:
I agree that just having a channel group within the current Discord would work better. If we have two it's just going to confuse people, and we're likely to again have people turning up in the development one going "why can't I ask questions here?"

I also think it's better to try to minimise the number of different places where AP questions are asked/answered - we already have Discuss, Facebook, Github and Discord, so I really think adding a second Discord would become a bit much. This also means users would see the dev calls etc running and that could help encourage them to join in development as well.

hendjoshsr71, 2022-07-16:
I disagree having separate servers "ArduPilot"  (for support & users)  & "ArduPilot-Development" would make it clear the purposes for each. Or at least allow us to point folks to the right server.  

It would also allow for channels that better align to the purpose of each user base. 
"Build" Categories, 
"First time setup",
"Crossfire"
"TX Setup" (I'd be in this one a lot!)
 etc?

We will though need to clean search engine links and our website so that people end up in the server they want. [Link](https://www.google.com/search?q=ardupilot%20discord)

Today, I chatted with one of the many folks who have come in recently and was confused by our setup.
And to thus my very scientific survey of 1 person /sarc :)

**How did she find the server?** Via google search "ardupilot discord"
**Top complaints:** Too many channels, server purpose not clear, 
**What was she trying to do?** First time vehicle setup and switch over from Betaflight & Inav
**What would she prefer to see a separate server or channels?** Server

timtuxworth, 2022-07-16:
> **How did she find the server?** Via google search "ardupilot discord" 

I think that says it all. 

She didn't search for "ArduPilot discord server that isn't the one that developers use, but the one for non developers that is only for support" - she searched for "ArduPilot discord".

https://www.youtube.com/watch?v=WboggjN_G-4

rmackay9, 2022-07-16:
I would also prefer we don't create another server.

auturgy, 2022-07-17:
I think a single server is better, but we introduce permissions for  write access to the dev channels. This provides the benefit of reducing noise for the devs, whilst maintaining visibility for the community.

IamPete1, 2022-09-04:
Our new user channel have certainty created a place for user support, dev channels are now a lot less busy. 

Maybe we can close this?

hendjoshsr71, 2022-09-04:
Seems good enough for now....

closed 2022-09-04 by hendjoshsr71