ArduPilot/ardupilot #21199: Partial parameter upload fails

booo, 2022-07-16:
## Bug report

**Issue details**

We would like to upload load a sub set of parameters to the plane. E.g. we would like to upload only servo configuration to the plane.

Right now partial parameter upload does not work for us. After partial parameter upload through QGC or mavproxy input from the remote is broken. E.g. none or only one channel works in radio calibration in QGC.

How to reproduce:

* reset all paramters to defaults
* reboot FC
* load partial parameters from file
* reboot
* open radio calibration in QGC

Here the parameters we try to use:

```
# Onboard parameters for standard searchwing vehicle
#
# Minimal searchwing configuration that can be merged with the upstream
# parameters from ArduPilot
#
# Stack: ArduPilot
# Vehicle: Fixed Wing
# Version: 4.0.3
# Git Revision: 1d77453
#
# Vehicle-Id Component-Id Name Value Type
#
1	1	ARMING_RUDDER	2	2
1	1	ARSPD_TYPE	0	2
1	1	BATT_AMP_OFFSET	-0.070000000298023224	9
1	1	BATT_AMP_PERVLT	18.339622497558593750	9
1	1	BRD_SAFETYENABLE	0	2
1	1	BRD_SAFETY_MASK	27	6
1	1	COMPASS_USE	1	2
1	1	COMPASS_USE2	0	2
1	1	COMPASS_USE3	0	2
1	1	INS_GYR_CAL	0	2
1	1	MIXING_GAIN	0.750000000000000000	9
1	1	RC3_REVERSED	1	2
1	1	SERIAL2_BAUD	9	6
1	1	SERIAL2_OPTIONS	0	4
1	1	SERIAL2_PROTOCOL	2	2
1	1	SERIAL4_BAUD	57	6
1	1	SERIAL4_OPTIONS	0	4
1	1	SERIAL4_PROTOCOL	10	2
1	1	SERIAL5_BAUD	921	6
1	1	SERIAL5_OPTIONS	0	4
1	1	SERIAL5_PROTOCOL	2	2
1	1	SERVO1_FUNCTION	4	2
1	1	SERVO1_MAX	2000	4
1	1	SERVO1_MIN	1000	4
1	1	SERVO1_REVERSED	1	2
1	1	SERVO1_TRIM	1570	4
1	1	SERVO2_FUNCTION	79	2
1	1	SERVO2_MAX	2000	4
1	1	SERVO2_MIN	1000	4
1	1	SERVO2_REVERSED	0	2
1	1	SERVO2_TRIM	1550	4
1	1	SERVO3_FUNCTION	70	2
1	1	SERVO3_MAX	2000	4
1	1	SERVO3_MIN	900	4
1	1	SERVO3_REVERSED	0	2
1	1	SERVO3_TRIM	900	4
1	1	SERVO4_FUNCTION	80	2
1	1	SERVO4_MAX	2000	4
1	1	SERVO4_MIN	1000	4
1	1	SERVO4_REVERSED	1	2
1	1	SERVO4_TRIM	1500	4
1	1	SERVO5_FUNCTION	4	2
1	1	SERVO5_MAX	2000	4
1	1	SERVO5_MIN	1000	4
1	1	SERVO5_REVERSED	1	2
1	1	SERVO5_TRIM	1500	4
1	1	SR2_ADSB	0	4
1	1	SR2_EXTRA1	5	4
1	1	SR2_EXTRA2	0	4
1	1	SR2_EXTRA3	1	4
1	1	SR2_EXT_STAT	1	4
1	1	SR2_PARAMS	0	4
1	1	SR2_POSITION	5	4
1	1	SR2_RAW_CTRL	0	4
1	1	SR2_RAW_SENS	0	4
1	1	SR2_RC_CHAN	0	4

```

**Version**

Version V4.2.2. of ArduPlane

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Hardware type**

We use a PixRacer

IamPete1, 2022-07-19:
Are your triggering RC failsafe? Can you upload a tlog and/or a disarmed bin log.

booo, 2022-07-21:
I uploaded some logs [here](https://downloads.borgers.it/tlogs/).

In [2022-07-21%2011-44-38-stick-movement-prior-to-parameter-load.tlog](https://downloads.borgers.it/tlogs/2022-07-21%2011-44-38-stick-movement-prior-to-parameter-load.tlog) you can see RC input. This is a system after a reset, with the serial configuration changed for the ExpressLRS module currently in use.

Immediately after parameter load you can still see input in [2022-07-21%2011-46-22-one-stick-still-moving-after-load.tlog](https://downloads.borgers.it/tlogs/2022-07-21%2011-46-22-one-stick-still-moving-after-load.tlog).

After a reboot you see only one channel changing in [2022-07-21%2011-48-47-no-movement-after-reboot.tlog](https://downloads.borgers.it/tlogs/2022-07-21%2011-48-47-no-movement-after-reboot.tlog).

How can I debug this further?

IamPete1, 2022-07-21:
You have your throttle input reversed and a low value of throttle failsafe, see:

https://ardupilot.org/plane/docs/rc-throw-trim.html#throttle-reversal

booo, 2022-07-21:
I added a proper RCMAP_* configuration. With the proper functions assigned to the channels the configuration works now. Thanks for the hint!

closed 2022-07-21 by booo