ArduPilot/ardupilot #21203: Simulation on Hardware

Lqs66, 2022-07-17:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_

## Bug report
When I follow the tutorial “[Simulation on Hardware — Dev documentation](https://ardupilot.org/dev/docs/sim-on-hardware.html#)” for hardware simulation, the connection to MissionPlanner keeps showing that the parameters are obtained, as shown below
[image]

**Issue details**
To implement the hardware emulation, I made two attempts.
1. I used the command ". /waf configure --board Pixhawk1 --extra-hwdef /cygdrive/e/ardupilot/libraries/SITL/examples/on-hardware/extra-hwdef-sitl-on-hw.dat - default-param /cygdrive/e/ardupilot/libraries/SITL/examples/on-hardware/default.param
. /waf copter" compiles successfully. Then I burned it to the pixhawk board and used missionplanner to connect to the flight controller and had the problem shown in the picture.
2. I added "env SIM_ENABLED 1" to the hwdef.dat file in the "ardupilot\libraries\AP_HAL_ChibiOS\hwdef\Pixhawk1" folder as described in the tutorial. ENABLED 1" and then use the command ". /waf configure --board pixhawk1 --extra-hwdef /cygdrive/e/ardupilot/libraries/AP_HAL_ChibiOS/hwdef/Pixhawk1/hwdef.dat
" to compile, then proceed to burn the firmware and connect to MissionPlanner, and still get the error shown above.

**Version**
ardupilot's master branch

**Platform**
 Copter

**Hardware type**
Pixhawk

Lqs66, 2022-07-18:
When I use QGC to connect to the earth station, after a long wait, I am prompted with the missing parameter "COMPASS_DEV_ID3"

auturgy, 2022-08-07:
@peterbarker