ArduPilot/ardupilot #21220: The default value of ATC_RAT_YAW_P in the firmware violates the documentation restrictions.

BlackJocker1995, 2022-07-19:
Defualt value of ATC_RAT_YAW_P is 0.09 in “ardupilot/Tools/autotest/default_params/copter.parm”.

But in [Document](https://ardupilot.org/copter/docs/parameters-Copter-stable-V4.2.1.html), the value of ATC_RAT_YAW_P should within (0.1, 2.5).

Is there something wrong with the code implementation or with the documentation?

IamPete1, 2022-07-19:
`ardupilot/Tools/autotest/default_params/copter.parm` is not the global defaults, it is just the values for the built in copter simulator. The default value of `ATC_RAT_YAW_P` is 0.18. As defined here: 

https://github.com/ArduPilot/ardupilot/blob/b59a2143ae42e77b6da02ab30ad452733a1e5b1f/libraries/AC_AttitudeControl/AC_AttitudeControl_Multi.h#L26

The ranges are only guidelines in anycase.

BlackJocker1995, 2022-07-19:
Thanks for the reply, I'll refer to the default from this file.

closed 2022-07-19 by BlackJocker1995