ArduPilot/ardupilot #21244: Automatic desired yaw with zero stick input

mbrymer, 2022-07-21:
## Bug report

**Issue details**

I'm flying in the simplest possible configuration in Stabilize mode with just IMU/gyro, no magnetometer/GPS, and have found that the vehicle has a strange tendency to yaw on its own. It doesn't happen every time, but it can occur spontaneously with zero yaw stick input or cause it to start rotating in the opposite direction of the current input.

I had a look at the logs and it appears that the yaw rate command to the inner loop is somehow getting an offset being added to it. See the plots below of throttle, yaw stick, desired and actual yaw rate on one and some yaw angles + yaw stick on the other. I did a simple hover test where I held the yaw stick at zero and found that within 3 seconds after takeoff(~20% throttle) it's requesting a 75 deg/s yaw rate despite zero yaw stick input. It appears the attitude controller is generating both a desired and actual yaw angle, and the spurious yaw rate request moves in line with the difference between the two. The interaction is unstable and moves back and forth repeatedly, it's also hitting angle wrap around which seems to not be helping. I'm not sure how it's getting a yaw angle...in absence of magnetometer the only option would be dead reckoning based on the gyro? There was some wind, so it could be getting set off by some small perturbations in yaw. Overall something seems to be causing it to try to do some sort of yaw angle control on top of yaw rate control, which is pretty confusing.

Log file has this test from 50 s onwards. I was initially trying to hover to log noises for notch filtering but switched after I got the yawing again. First 30 seconds are zero yaw stick input as seen in RC4. I eventually couldn't stomach it anymore and took over.
[image]
[image]

I'm new to ArduPilot, but have flown a number of quads and never experienced something like this with pilot inputs being overridden in a manual mode. It's pretty dangerous, I've already had a few flips when it gets so bad that you can't count on your roll/pitch being in the direction you're requesting. I'm going to get a GPS + magnetometer to see if it helps sidestep the issue, not sure if this is just an atypical configuration I'm running.

As a troubleshooting step I've tried disabling this by setting ATC_ANG_YAW_P to a small value(1E-10) and overriding the parameter out of range arm condition, but it appears to be getting clipped back to the minimum value because the issue remains.

Is there some other workaround that might help work around this yaw behaviour?

**Version**
4.2.1

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad, 5" props

**Hardware type**
Holybro Kakute F7 V1.5 FC, no GPS/compass, IMU only

**Logs**
Full log here: https://drive.google.com/file/d/1bMpKC3KH49HbwDzO8pSyz6Et_hxDj8-G/view?usp=sharing

rmackay9, 2022-07-21:
I have not yet looked at the logs but my guess from the description above is that this is caused by an estimation error.  The EKF's yaw and yaw rate are inaccurate because there is go compass (or any other absolute heading source) to counter act the gyro rate drift.

IamPete1, 2022-07-23:
The yaw outputs are dominating roll and pitch so your motors may be twisted or rotating the wrong way. From the attached log I do not see any autopilot issue.

Maybe start a thread over on the forum with some more details of the hardware.

closed 2022-07-23 by IamPete1