ArduPilot/ardupilot #21246: ADSB_TYPE 0 causes MavFTP errors

BKims, 2022-07-21:
## Link to discusison
https://discuss.ardupilot.org/t/lua-script-upload-through-mavftp-error/86526/2

## Bug report

**Issue details**
When param ADSB_TYPE is set to 0, MavFTP does not work correctly. It can be seen in the command line of MP, and when uploading scripts.
Following error can be seen when trying to upload scripts 
[image]

**Version**
AC 4.2.2 stable

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Cube Orange

**Logs**
no logs

IamPete1, 2022-07-21:
I cant replicate the issue. Can you give some more info on the setup? How are you connecting? Is it the same over USB? What params are set?

It could be a mission planner issue, which version are you using? Is it the same on the latest beta?

magicrub, 2022-07-21:
I can't see how that's related. Please provide:
- autopilot hardware
- dataflash Log (.bin), you can post it here in github
- param file, post here in github
- connection type (USB, serial, radio...)

rmackay9, 2022-07-21:
Yes, I think it is MP related so I'm going to close this.  We can re-open if it turns out to be a flight code issue.

closed 2022-07-21 by rmackay9

BKims, 2022-07-22:
> 

-As said in the bug report, Hardware is Cube Orange
-I cannot provide any logs since this does not happen when the FC is already connected,  rather when connecting to the Cube (stuck on Check MAVFTP) or when uploading Lua Scripts via MavFTP (error in my post)
-param file attached to this reply
-connection type does not matter, had the problem via USB and UDPCI (via Herelink on same Wifi)
-MP version is 1.3.77

[20220720 Q6v3-Rev1 - Premier vol sans PID.zip](https://github.com/ArduPilot/ardupilot/files/9165303/20220720.Q6v3-Rev1.-.Premier.vol.sans.PID.zip)
)

rmackay9, 2022-07-22:
To add credence to @BKims's report [we've had a similar report here during 4.2 testing](https://discuss.ardupilot.org/t/copter-4-2-0-rc2-released-for-beta-testing/84818/3).

WickedShell, 2022-07-22:
I have a plane here with 4.2 that is nigh unusable with Mission Planner due to not being able to download parameters over FTP. It eventually times out and falls back to individual parameter download, which took about 10 minutes to complete. Both MAVProxy and my own GCS happily connected without any problems on the same link. (My GCS didn't use FTP at all).

It's also worth noting that MissionPlanner worked fine when connecting over USB instead of a radio.

Details:
CubeOrange
ArduPlane 4.2
Radio: RFD900X running 3.57, not MAVLink aware, no flow control
Stable Mission Planner download as available July 21st
Radio on serial 1
`ADSB_TYPE` 0

hendjoshsr71, 2022-07-22:
Hmm this may be also an issue I saw with partner I was working with. Over their RFDs we nearly always had to wait for MAVFTP to fail and fall back.

geofrancis, 2022-07-22:
I get mavftp errors when running ardurover 4.3 on a pixhawk 1, I never got to the bottom of it, I was having to restore default parameters in order to get mavftp working again so I could copy over lua scripts then restore my parameters back. I havent had time to change each setting one by one to figure out what specifically was causing it, I just assumed it was due to lack of memory or something due to having a lot of things enabled.

WickedShell, 2022-07-22:
> I just assumed it was due to lack of memory or something due to having a lot of things enabled.

On the system I'm reporting from I have a bit over 577,000 bytes free, so I doubted that was my problem.

rmackay9, 2022-07-23:
FYI @meee1.